OrcaSlicer/OrcaSlicer #3242: Can't create a new filament

tlhintoq, 2023-12-22:
### OrcaSlicer Version

1.8 and newer through 1.9 dev 23

### OS version

Windows10x64

### Printer

Any

### How to reproduce

1. From main screen
2. Filament - settings gear icon
3. Try to create a new custom filament

### Actual results

You can't select from the drop down as it is disabled.
There's no way to select an existing material or printer.
You get an error popup that you have not  yet selected a printer or material.
[image]

### Expected results

I should be able to make/add TPU as a material

### Project file & Debug log uploads

No crash.  No project required.  Happens at any point

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2023-12-22:
Oh - also, the create new filament dialog pops up on the primary monitor instead of the monitor where the program is.

[image]

tlhintoq, 2023-12-30:
Still an issue on 1.9.0 beta

tlhintoq, 2024-01-17:
Keep alive - Still an issue in 2.0 developer/nightly

robert-stevenson-1, 2024-01-18:
I've also experienced the same issue.
(Interested in fix, hence trying to get some engagement here)

Senayna, 2024-01-24:
Same here. Tried to add some ABS+ by eSUN for my Raise3D Pro3 Plus. Not able to, for the same reason as stated above.

huntervaneck, 2024-02-14:
I have the same issue. on 1.9 and 2.0. Unable to add any custom filament profiles.

66mkiqdv, 2024-03-06:
Same for me too.

[image]

imagesurgery, 2024-03-11:
Same or similar issue where I cannot add a custom filament to Orcaslicer 1.9.1 on macOS Sonoma 14.3 (installed via home-brew):
https://github.com/SoftFever/OrcaSlicer/assets/4323784/a8cf24e8-1f3a-4f3d-a118-b98d42d8df7c

Seems to be the same with any combination of system or user presets, for any printer; only way to proceed is to select the wrong filament type in the dropdown, but I cannot figure out how to change this afterwards to the correct filament.

samv, 2024-04-08:
I'm just trying to select ASA for my printer.  I was able to add it for "Voron" but not "Sovol SV06 Plus".  
[image]

tlhintoq, 2024-04-15:
Keep alive.  Problem still exists in current 2.0 nightly as of 15apr2024

Philomorph, 2024-04-19:
I don't know if it's helpful to say "me too" but I run into this whenever I try to add a new filament for my AD5M. I end up having to select a "system" filament with the Flashforge name and then just edit it and rename on save.

samv, 2024-04-20:
FWIW, I found a work around for this.  I can edit a profile for a filament type it thinks I can print, such as ABS, and then edit it.  In that view, I can enter whatever I like,

[image]

Maybe one day I'll have a beltless machine with CNC parts that can do filaments like that.  Until now, I can only dream using this filament entry dialog.

Madin5, 2024-04-28:
Same problem. Can't add TPU profile.
[image]

TheIncgi, 2024-05-09:
Found another workaround based on [this reddit post](https://www.reddit.com/r/3Dprinting/comments/1bj1s6l/orcaslicer_cannot_add_filament_see_image_captions/)
My selected printer was `Elegoo Neptune 4 Max`, I temporary switched over to a generic `klipper` printer and was able to create a TPU preset after that.

**To get it to show up under my existing printer preset:**
I went into `AppData\Roaming\OrcaSlicer\user\default\filament\base` to tweak the json for the new profile a bit
I edited these:
 - `compatible_printers`
 - `filament_settings_id`
 - `name`

I used another preset I already had in as reference for the `compatible_printers`
I also renamed the files to match. (`.info` and `.json`)

tlhintoq, 2024-05-10:
Bot nudge.

WillingSpy, 2024-05-17:
The current workaround is to create it as a PLA/ PETG.

It seems only some filament types don't have the printer option for some reason.

using 2.0.0

dockrbyter, 2024-06-03:
Same problem here with 2.0.0.... Will this be fixed in the upcoming version? 🤔

tlhintoq, 2024-06-04:
Here's what happens for me in 2.1.0-beta

Try to make a new material.
[image]
Click `Create New` on the next dialog

The next popup jumps location way over to another monitor and half off the screen
[image]

It offers to make a material based on an existing material for an existing machine.
[image]

Then I'm stuck because I have to make up a serial number for a material.  Huh?  "BlahBlah black PETG" has to have its own serial number?  Anyway, no way to enter this, so stuck.
[image]

So I canceled and started over.  Again and again started over.  Thinking I could fill out the box in reverse order or something before it flew off the screen, just so long as I could click [done] when done.
[image]
[image]

But in the end changing just about anything like the maker, or material type makes the dialog fly off.
Then there is the whole inheritance lineage to mess with.  Seems a new material is going to inherit from something different for every machine.  `PETG-Generic` on one `Generic-PETG` on another `My PETG`... `Standard PETG`.

All of these materials at some point... at some point there had to be a top level material that didn't inherit from others, right?  Something had to be first.  So if I'm making a **NEW** material, can't it be a top level material that doesn't inherit from anyone?  Can't my new material be genuinely new?
[image]

dcchillin46, 2024-06-04:
Just dropping by to say that as of 6/4 in 2.1 nightly build im still having this issue. I can edit an existing profile and save it but thats a bit of a hassle.

dockrbyter, 2024-06-10:
Why isn't here any reaction from @SoftFever? It's not that this topic is just few hours old 🤷🙎

RayzenOri, 2024-06-25:
OrcaSlicer 2.1 can't add filament because the program doesn't display any printer.
[image]

obertini78, 2024-06-26:
I can confirm the same issue as with 2.1.0

MuteKnee3D, 2024-06-27:
My current solution to this problem (Orca 2.1.1):
Choose an existing filament (i.e. "generic PLA")
Edit the filament by choosing the little pencil and paper icon next to the filament.
Change the color and then click the "save" (floppy disk) icon.
Name the filament anything you like (i.e. SUNLU Transparent)
Run your calibrations and edit your new filament profile.

pkysela, 2024-06-27:
I found a work around, open OrcaSlicer 1.9.1, add your new custom filament, open OrcaSlicer 2.1.1 and hey presto it's in the list ready for fine tuning. (works on Windows)
I assume OrcaSlicer stores its data is user's AppData? It would be good to know, in case I need to switch computers in the future.

MuteKnee3D, 2024-06-27:
@pkysela
> I found a work around, open OrcaSlicer 1.9.1, add your new custom filament, open OrcaSlicer 2.1.1 and hey presto it's in the list ready for fine tuning. (works on Windows) I assume OrcaSlicer stores [...]

Yes
 C:\Users\"yourname"\AppData\Roaming\OrcaSlicer\user\default
3 folders = filament, machine, process

NeWFieKaNaTa, 2024-07-26:
Same issue here. Posting for more attention.

jotape26, 2024-08-13:
Found a workaround that worked for me on Mac:

- I've created a custom profile using a printer that I can select under "Copy Current Filament Preset", in my case I used Sovol SV06 0.4 Nozzle.
- This created a .info and a .json file under `/Users/<user name>/Library/Application Support/OrcaSlicer/`
- Rename both files. Don't know if this affects anything but I renamed both files to use `Voolt3D PLA Velvet @Creality K1 (0.4 nozzle)`
- On the .json file, I've added the compatible_printer that I use: Creality K1 (0.4 Nozzle) [I've added the parenthesis as well in case it needs to match the name of the printer preset]
- Restart Orca and verify that it worked.

Is not an elegant solution and it doesn't even copy the Manufacturer's preset but at least it something you can expand upon.

AlejoMSP, 2024-08-17:
so, is this issue being looked after?
I'm ready to jump back into CURA if this si not resolved, seems pretty easy fix

tlhintoq, 2024-08-19:
> so, is this issue being looked after? I'm ready to jump back into CURA if this si not resolved, seems pretty easy fix
After all this time - I assume its not.

**### Not worth the lobotomy of going back to Cura.**  

- Pick an existing filament.
- Make changes you want
- Save as new filament

Not awesome but gets the job done

kenni, 2024-08-19:
> **### Not worth the lobotomy of going back to Cura.**
> 
> * Pick an existing filament.
> * Make changes you want
> * Save as new filament
> 
> Not awesome but gets the job done

Unfortunately, it does not get the job done. The end result will technically end up as two different types of filament profiles. If you have for example Bambu Lab printers with AMS, you need to create new custom profiles and not copy an existing, in order to sync the profile to your printers. From the Bambu Lab wiki:

"Compared to Method one, Method two is the fastest way to quickly modify and create a new preset based on an existing filament preset. However, the filament created in Method two cannot become a system-level filament preset like the customed filament created in Method one, so it cannot be synced to the printer. It is also not possible to select that filament for an AMS slot on Studio's ‘Device’ page and sync it to the filament list."

dbloemhard, 2024-08-20:
So many people having this issue and for so long... But not a peep from the developers. Doesnt make you feel confident that this software will get fixes for other issues.

tlhintoq, 2024-08-20:
Its has been... and you're right about confidence.
I *suspect* a lot of the silence comes from the fact that Orca is a branch from another program.  I *suspect* that the source of the problem is in the parent program Orca inherits from.  And thus rewriting much of that inherited problem meaning opening a BIG can of worms and ongoing maintenance.
But if that's the case then Orca development just needs to be transparant about that.  Respond to all the threads & tickets with honesty.  Just say:
> The material handling and management comes from Prusa slicer base code that we inherit and as such we're not going to try to re-write that entire subsystem as it is beyond what we have the developmental resources for at present.
or something like that.

dbloemhard, 2024-08-20:
I tried the workaround suggested (manually adding my filament by copying
another one) and it was one of the worst tpu prints I've done. Obviously
missed a few important settings and I'm not experienced enough to know
which ones they are (UI seems to hide basic stuff like retraction and print
speed or I am too dumb to find it). I just installed Cura Slicer and it has
good profiles for that and all other filaments out of the box. Problem
solved!

timnolte, 2024-08-21:
> After all this time - I assume its not.
>
> **### Not worth the lobotomy of going back to Cura.**
>
> - Pick an existing filament.
> - Make changes you want
> - Save as new filament
>
> Not awesome but gets [...]

Yeah, this is essentially what I've ended up doing. Which works fine for the most part but these are just custom user presets tied to a specific base filament profile and it doesn't show that I have any custom filaments in Orca. 🤷‍♂️

samv, 2024-08-23:
To be honest, the whole system of filament profile management in Orca Slicer needs an overhaul.  What's asinine to me, even without this bug, is that:
1. my workaround above does not work when you move to multiple printers; these "forked" filaments created by saving a filament with a changed material field are basically "second class filaments" that cannot be easily used by other printers
2. when coping a filament profile from one printer or nozzle to another, it requires that there be an existing profile _of the same type_ on the _destination_ printer.  IME there might be _some_ settings which are custom to the printer, but honestly, the print settings for filament are mostly a property of the filament, and any adjustments for each printer's nozzle like flow or thermistor for temperatures are likely to be systematic, at least at a first pass.  It should at least copy across the commonly listed filament properties like the recommended temperature range.

To solve this, for a start each property in the filament properties dialog could show where it is being inherited/set from.  When creating a new preset for a filament, it could allow copying all the properties from the source printer, and if there are profiles for similar materials on the destination printer, the properties could be compared and merged.  In fact if there are _many_ filaments on the target printer of the correct type, it could even provide some kind of statistics: eg: "temperature ranges for ABS on this printer: 230 - 280" / "for other ABS of the same manufacturer: 230 - 260." then "temperature range on source printer: 230 - 260".  "Use 230 - 260?"

I want to see, instead of rejections that there is no profile for that type of filament yet, warnings like "No PPS-CF filaments defined on this printer yet.  Create a new profile from scratch?"

lyr0x, 2024-08-24:
MacOS 14.6.1 same problem with Orca Slicer 2.1.1. Filament created with Klipper can't be used for other printers. Please rework the filament section

PM3Delta, 2024-09-03:
Yes, please fix this - same problems as described. I have a modded Sovol SV01, which prints Nylon, PC, ABSplus and I cant add the pre-created profiles from the respective filament makers (ESun, Polymaker, Taulmann etc.), but have to use the above mentioned unsatisfactory workaround by e.g. selecting PLA, then enter all the data manually and save it under a new profile name.

ifihaffto, 2024-09-08:
Gosh.

I have just dipped my toe into Orca slicer (coming from Cura) and being utterly bamboozled at setup :(
Now, at this point, I understand that I know NOTHING.

I have 2.1.1 installed on my Mac and I can't create/add TPU filament for my Ender printers.
I have to create a generic printer, and generic TPU filament, but I can't _**call**_ it 'Generic TPU' Filament because that name is already taken - I just can't actually apply it.

1. Why no Creality TPU?
2. Also can't create filament because it says I have no printer selected. (My installed printers are Ender 3 Pro, and Ender 3 S1 Pro)

timnolte, 2024-09-08:
Yeah, something was only slightly improved as now it doesn't crash when trying to add a custom filament but then I too just get the no printer selected error which basically prevents the custom filament feature from being usable.

SoftFever, 2024-09-09:
I can't repro the issue.
Can you provide more info?
https://github.com/user-attachments/assets/f2992e63-309d-4bf5-9738-1a4685ecef55

kenni, 2024-09-09:
> I can't repro the issue. Can you provide more info?
> 
>  custom.filament.mp4

@SoftFever Could it be that you have a Bambu Lab printer configured? Initially, I had a clean install of Orca Slicer and setup an Elegoo Neptune 4 printer in Orca Slicer - and I had this exact issue. Later on when I got a Bambu Lab P1S setup in Orca Slicer, I noticed that the filament features were fully working for the Bambu Lab printer, but still not for the Elegoo printer.

timnolte, 2024-09-09:
I also don't have, and never had, a Bambu printer. I have a FlashForge Adventurer 5M.

timnolte, 2024-09-09:
> I can't repro the issue.
> Can you provide more info?
> 
> https://github.com/user-attachments/assets/35e638f3-42bb-445a-8711-fd5543413c9d
> 
> 
> 

When I have that screen I don't even have a selection for choosing a printer.

numbermess, 2024-09-09:
Mine is the same way. I can't select the Neptune printer when creating the new filament because only the Bambu printers appear in the list:

[image]

timnolte, 2024-09-09:
> I can't repro the issue. Can you provide more info?
> 
>  custom.filament.mp4

@SoftFever I noticed in your video you selected some filament called "My Generic TPU" which seems like you started with some custom filament to create your new custom filament. What happens when you select a specific vendor filament profile? I was using the FlashForge filament profile not a custom one(that I can't create anyways).

SoftFever, 2024-09-11:
There is currently a limitation where you can't create a material type (e.g., ABS-GF) if the selected printer profile doesn't have this type of filament defined.

Meanwhile, for the filament types supported by your machine, can you check if the copy option works, as shown below:
https://github.com/user-attachments/assets/41ec9fb7-6a04-47e9-bab5-ba5db465dc7e

If it's not working
Can you please report in this thread:
1. what printer
2. what filament type
if you are able to record the screen, it would be very helpful!

kenni, 2024-09-11:
> 1. what printer
> 2. what filament type
>    if you are able to record the screen, it would be very helpful!

@SoftFever Here's a screen recording of me not being able to add Prusament PLA.

1. Printer: Elegoo Neptune 4 Plus.
2. Filament: Any, but in the screen recording I'm trying to add Prusament PLA.

Link to screen recording:
https://drive.google.com/file/d/1YkwwcYtISkoprgkzGZYxt-ZpV2dTxlMS/view?usp=sharing

dbloemhard, 2024-09-11:
Printer: Creality Ender 3 V2
Filament: Sunlu TPU
Scren recording attached. My printer doesnt even show up in "selected printer" despite my printer being selected

https://github.com/user-attachments/assets/aec74c6f-f590-40c9-84f9-9907a645c365

SoftFever, 2024-09-11:
> Printer: Creality Ender 3 V2 Filament: Sunlu TPU Scren recording attached. My printer doesnt even show up in "selected printer" despite my printer being selected
> 
>  Screen.Recording.mp4

thanks for the video.
this falling into this:

> There is currently a limitation where you can't create a material type (e.g., ABS-GF) if the selected printer profile doesn't have this type of filament defined.

This printer's profile has incomplete filament support.
It doesn't support TPU

SoftFever, 2024-09-11:
> @SoftFever Here's a screen recording of me not being able to add Prusament PLA.
>
> 1. Printer: Elegoo Neptune 4 Plus.
> 2. Filament: Any, but in the screen recording I'm trying to add Prusament PLA. [...]

thanks.
Gonna look into it

dbloemhard, 2024-09-12:
> This printer's profile has incomplete filament support.
> It doesn't support TPU

Ok so how can we edit this or fix this? I created a new printer, updated printable space and then selected import preset, selecting the Ender 3 V2. This then allowed me to select all filament preset templates, including the generic TPU filament.

So now i have Generic TPU available from the filament drop down, but still when i go to add a new custom filament, my printer is not shown in the add custom filament preset when selecting a TPU filament.

geekredux, 2024-09-12:
Still happening in Orca Slicer 2.1.1 on Windows 10. Printer is Neptune 4 Pro.

Filament management is really terrible. Among other things, it feels like the adding filament process is deceptive. You can choose to add filament, then select "Custom Filaments", then you get to click a button called "Create New". But, really, this button is a lie. You can't "create new", only "Create based on current filament" (which is often a completely different brand from what I'm trying to add), or "copy current filament preset" (which is apparently but not obviously different from "create based on current filament" somehow).

And regardless, either choice still gives that same "you haven't chosen a printer" error.

Anyhoo, effort and error screengrab attached:
[image]

PS: Why are filament properties inherited? Sure, copy properties from existing profiles, but create a new standalone profile. Not being able to delete a filament profile because a different profile was created based on it is really aggravating.

markaudacity, 2024-10-04:
Still having this issue in Orca 2.1.1 in Win 11.

maniacrush, 2024-10-10:
Orca Slicer 2.1.1, Win 11
I was able to add new TPU filament for my Creality Ender 3 S1 Pro with the following workaround:
1. Added my printer to the "compatible_printers" list in the filament config file(in my case "Creality Ender-3 S1 Pro 0.4 nozzle"):
c:\Users\user_name\AppData\Roaming\OrcaSlicer\system\Creality\filament\Creality Generic TPU.json
2. Added "Creality Generic TPU" to the "default_filament_profile" of my printer config(not sure if this step is really required):
c:\Users\user_name\AppData\Roaming\OrcaSlicer\system\Creality\machine\Creality Ender-3 S1 Pro 0.4 nozzle.json
After that the "Creality Generic TPU" preset appeared in dropdown and I was able to select my printer.
[image]
I suppose that the problem is in the config files, filament configs are simply missing printers that support this filament type.

stefansjs, 2024-10-12:
Can we get a workaround? I've been trying to create an ABS profile (which is not an unusual filament) by copying .json files from `C:\Users\sulli\AppData\Roaming\OrcaSlicer\system\BBL\filament`. I created a new file called `Bambu ABS @BBL A1M.json` and edited the "setting_id" and "compatible_printers" fields:

```json
{
    "type": "filament",
    "name": "Bambu ABS @BBL A1M",
    "inherits": "Bambu ABS @base",
    "from": "system",
    "setting_id": "GFSB00_17",
    "instantiation": "true",
    "fan_max_speed": [
        "20"
    ],
    "filament_max_volumetric_speed": [
        "16"
    ],
    "hot_plate_temp": [
        "80"
    ],
    "hot_plate_temp_initial_layer": [
        "80"
    ],
    "reduce_fan_stop_start_freq": [
        "0"
    ],
    "slow_down_layer_time": [
        "12"
    ],
    "textured_plate_temp": [
        "80"
    ],
    "textured_plate_temp_initial_layer": [
        "80"
    ],
    "compatible_printers": [
        "Bambu Lab A1 mini 0.4 nozzle",
        "Bambu Lab A1 mini 0.6 nozzle",
        "Bambu Lab A1 mini 0.8 nozzle"
    ]
}
```

 I then edited `C:\Users\sulli\AppData\Roaming\OrcaSlicer\system\BBL.json` and added a reference to the new filament profile by adding: 
```json
        {
            "name": "Bambu ABS @BBL A1M",
            "sub_path": "filament/Bambu ABS @BBL A1M.json"
        },
```
to the `filaments` list right under 
```json

        {
            "name": "Bambu ABS @BBL A1",
            "sub_path": "filament/Bambu ABS @BBL A1.json"
        },
```

Even after doing all this I still don't see the ability to select ABS for a BBL A1 mini. It seems like there's still some configuration elsewhere in the slicer that indicates which "base" presets are available for a given printer.

It would be really great to know how to tell OrcaSlicer that a given printer does support a given filament. Or to add a new filament type from another printer.

https://github.com/user-attachments/assets/700b1572-06fb-46b2-bf61-db5b90ea7572

aqkhana2002, 2024-10-13:
I HAVE THE SAME ISSUE, the worst program ever.

stefansjs, 2024-10-14:
It's free. Don't be a jerk to somebody who is literally volunteering their time to give you a professional class program for your hobby.

tlhintoq, 2024-10-14:
> I HAVE THE SAME ISSUE, the worst program ever. how come this program allowed to be in production ?

If you don't like the show, change the channel.
[And if you don't grasp that - if you don't like this slicer you're not required to use it - use one of the countless others out there.]

malventano, 2024-10-19:
> 1. what printer: X1 Carbon
> 2. what filament type: [ASA-CF
](https://github.com/SoftFever/OrcaSlicer/issues/7030)

X1 Carbon is unable to add any ASA-CF, even Bambu brand (also causes crashes on device tab if ASA-CF is already mounted as an external spool). Can't add as a custom filament per @Azio-Pantheon's workaround, since the MyKlipper profile does not support ASA-CF either.

Tried manually copying over the updated json/profiles from recent Bambu Studio, but Orca does not appear to use the new configs. Can't get ASA-CF to show up as a system filament, etc.

Atrox101, 2024-10-20:
I was able to solve my problem.
I set up the material on another printer under Custom Filaments and then I went into Custom Filaments and simply clicked on "Add Preset". There I was then able to add the missing printer with the material.

[56 more replies]

PR #10362 (2025-08-11): Fix inability to create custom filament due to hidden printer selection checkbox
This PR resolves an issue where the Create Filament Preset dialog did not display the printer selection checkbox, preventing users from creating a custom filament.

The root cause was that the main scrolled panel was added to m_main_sizer with proportion=0, preventing vertical expansion and hiding the checkbox. Updating to proportion=1 ensures the panel expands to fill the space, making all controls visible.

Fixes #3242, #6719

# Changes

- Changed `m_main_sizer->Add(..., proportion=0)` → `proportion=1` in the Cr [... cut]
closed 2025-08-11 by SoftFever