OrcaSlicer/OrcaSlicer #3250: [Feature Request] Add Brim Blocker

jordanlambrecht, 2023-12-23:
It would be extremely useful to have a way to prevent a brim in certain areas, especially tight nooks and crannies where it's hard to get a deburring tool into.

Eldenroot, 2023-12-23:
https://github.com/SoftFever/OrcaSlicer/issues/3131

DVSVIDEO, 2023-12-26:
I recommend using Mouse Ear in such cases.

Eldenroot, 2024-03-26:
B

notsaeegavas, 2024-04-02:
> I recommend using Mouse Ear in such cases.

This didn't solve my issue as I used mouse ear brims and it sliced brims on an inner wall where I didn't want them. A brim blocker would be a great feature.

closed 2024-07-10 by github-actions[bot]

Snuff1eupagus, 2024-08-12:
@jordanlambrecht 

Brim Blocking,

https://www.youtube.com/watch?v=6CRqNk_JUj4

Hopefully this will do the trick for you.

MJ3Ds, 2025-01-04:
@Snuff1eupagus link is dead got a new one?