OrcaSlicer/OrcaSlicer #3254: What happened to only_retract_when_crossing_perimeters?

evranch, 2023-12-23:
**Is your feature request related to a problem? Please describe.**
In Cura this is called "combing", in the upstream slicers like PrusaSlicer this is "only_retract_when_crossing_perimeters". It lets you cut the retraction count down tremendously and prevent filament chewing in Bowden systems. OrcaSlicer has "Avoid crossing walls" which avoids crossing walls, yes, but it doesn't stop the retractions.

This feature doesn't seem to exist in Orca and doesn't even seem to be present in the GCode generator, though there are remnants remaining in configuration structures. Was it removed for some reason?

I've noticed Orca seems to generally consider retraction to be a free action where Cura and PrusaSlicer give a variety of options to reduce it.

**Which printers will be beneficial to this feature**
Bowden tube printers, particularly large/long tube systems and ones with light heads / high travel speeds like my highly modified TronXY X5SA. I would usually rather travel a long comb path than take a retraction, due to chewing, long retraction lengths and repriming issues at high print speeds.

**Describe the solution you'd like**
Wipe and travel into infill or comb back along perimeters instead of retracting when this mode is enabled.

**Describe alternatives you've considered**
Switching back to Cura or Prusaslicer after several lost prints due to chewed filament.

**Additional context**
These retractions are not necessary as there is a path through the infill, and in fact most of these perimeters are adjoining.
[image]

igiannakas, 2023-12-26:
Its called reduce infill retractions and setting this to on is the same as prusa feature.

evranch, 2023-12-30:
Thanks, I had this turned on as well but it's good to know what it actually represents.

I just realized that when slicing in vase mode (as this print was) these settings appear to be silently ignored, resulting in increased retractions in the base layers compared to a regular print.

PixelKaje, 2023-12-31:
I would also like Combing added to Orca Slicer. It's not so much about retraction as it is traveling over a printed surface. Reduce Infill Retractions helps a bit, but does not solve the issue. I have travel moves over an infill surface that will create a line that is still visible after the top layer has printed. It more or less ruins the print. 

There are two reasons this happens - one, it travels over a printed surface. With Combing, you have the ability to restrict travel to ONLY perimeters, which would solve the issue.

Two, I've noticed that at the end of each layer, the nozzle will "dab" various spots around the printed surface (what I'm assuming is "tiny gaps", but changing that setting has zero effect). When it does this, there are no retractions. When going around the print to place these tiny dots, retractions are ignored, as are travel paths - the nozzle essentially goes wherever it wants to, and ignores all other settings. If there was a setting that would allow the print to ignore or skip filling in those tiny spots, which have zero effect on the quality of the printed surface, that would also fix this issue. On PrusaSlicer this setting is called "Fill gaps". Deselecting that checkbox prevents the printer from dropping those little dots. In Orca Slicer's Filter out tiny gaps, nothing I input affects the actual print.

igiannakas, 2023-12-31:
There is an option to avoid crossing perimeters - which sounds like what you need? That turned on and reduce infill retractions turned off = combing?

PixelKaje, 2023-12-31:
Not quite. I don't need to avoid crossing perimeters (even though I have that setting enabled). It needs to avoid crossing infill and travel ONLY along the perimeter/walls.

PixelKaje, 2023-12-31:
[image]
This is what's needed.

evranch, 2023-12-31:
I agree that both Cura's combing algorithm and ability to avoid parts are superior. I should have mentioned that Combing can be set to stay in perimeters. As well as top layers, this can really improve a first layer by avoiding creating tiny strings that get baked in.

Perimeter combing with a long Max Comb Distance can retrace long paths along a perimeter to avoid a retraction and travel move. It will slide around to hit those final tiny dots without either chewing filament or stringing, which can give a far superior finish as well as better print times if you have a well tuned motion system.

This type of combing absolutely shines with flexibles in a long Bowden system as well, because minimizing retractions to near zero avoids both the oozing and low nozzle pressure on restart.

lizardjazz1, 2024-01-11:
Guys, I subscribe to every word you say, this is really a very important function

[image]

lijxisomer, 2024-03-04:
Just putting in my support to this feature. Combing mode is one of the best and must-have features introduced by Cura. It almost completely removes stringing problems by restricting moves only to along perimeter walls and within infill.

ramosglauco, 2024-03-23:
Hi. Combing mode is perfect. However, it is extremely important to also have the option of the nozzle not traveling through the perimeters, this way the problem of bloobs is avoided.

MAJCcustoms, 2024-04-30:
I support better path planning features like combing. Extremely valuable in specific situations.

axtox, 2026-11-07:
well, I think this feature is still wanted, is there anything changed in 2.5.0 Orca? I am struggling with my bowden setup