OrcaSlicer/OrcaSlicer #3255: "Only one wall on top surfaces" can cause artifacts for small features on top of the surface (lettering issue)

psiberfunk, 2023-12-23:
### OrcaSlicer Version

1.8.1

### OS version

Win 11

### Additional system information

N/A

### Printer

Bambu X1C

### How to reproduce

Open attached file, slice.  Toggle "Only one wall on top surfaces", reslice.  Observe different behavior inside the outer perimeter.
[OrcaSlicer_1WallOnTopLetteringBug.zip](https://github.com/SoftFever/OrcaSlicer/files/13760218/OrcaSlicer_1WallOnTopLetteringBug.zip)

### Actual results

When "Only one wall on top surfaces" is on, it very often causes the top surface of a print with lettering on top of it to be printed in a non-monotonic fashion in small little sections, which causes visual defects in the print.  It appears to be printing "inner walls" where it should be printing a top surface (see line type in this screenshot), and areas circled in red marker.

[image]

### Expected results

 "Only one wall on top surfaces" should not be causing issues like this on the top surface.  As soon as you turn it off, the behavior returns to the expected/desired behavior:

[image]

### Project file & Debug log uploads

Log file N/A, not a crash/graphical/UX issue.. this is a slicing engine issue.
[OrcaSlicer_1WallOnTopLetteringBug.zip](https://github.com/SoftFever/OrcaSlicer/files/13760228/OrcaSlicer_1WallOnTopLetteringBug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

TechDude23, 2023-12-24:
Hello,

Set one wall threshold to 0 

The setting on Bambu Studio is 100%

SoftFever, 2023-12-24:
@psiberfunk 
You can play around with the "One wall threshold" to get the desired result.
The parameter is added for such edge cases ;) 
In some models we want to increased the threshold.
For this specific case, you might want to reduce it to 50%.
[image]

HakunMatat4, 2023-12-24:
Hmmm I have crossed this issue on the past, opened a ticket and I was told to disabled one wall on top surface for some models.
Not great but it is what I have been doing ever since.
Oh well.....

SoftFever, 2023-12-24:
> Hmmm I have crossed this issue on the past, opened a ticket and I was told to disabled one wall on top surface for some models. Not great but it is what I have been doing ever since. Oh well.....
> 
> EDIT: Voron profiles also have that threshold set to 300% lmao

The famous spiral learning curve ;)

psiberfunk, 2023-12-24:
So , why , on BambuSlicer , which also has this feature , is this not an issue with the default settings ?

Noisyfox, 2023-12-27:
Because they have different default settings.

psiberfunk, 2023-12-27:
Might I suggest that their defaults are better if they don’t produce this issue ?

psiberfunk, 2024-01-07:
I would point out @SoftFever that this bug happens with default settings with even the orca calibration prints.. which suggests maybe the defaults are not that great ?  See attached temp tower where the text has this issue.
[image]

closed 2024-03-16 by SoftFever