OrcaSlicer/OrcaSlicer #3258: [Feature Request] Windows 11 ARM64 build

eblieb, 2023-12-24:
Any chance we can get a native Windows 11 Arm64 build for Orca Slicer?

giveen, 2023-12-27:
You could always try building it yourself?

in the build script:
https://github.com/SoftFever/OrcaSlicer/blob/main/build_release_vs2022.bat

```
echo cmake .. -G "Visual Studio 17 2022" -A x64 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="%DEPS%/usr/local" -DCMAKE_INSTALL_PREFIX="./OrcaSlicer" -DCMAKE_BUILD_TYPE=Release
cmake .. -G "Visual Studio 17 2022" -A x64 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="%DEPS%/usr/local" -DCMAKE_INSTALL_PREFIX="./OrcaSlicer" -DCMAKE_BUILD_TYPE=Release -DWIN10SDK_PATH="C:/Program Files (x86)/Windows Kits/10/Include/10.0.22000.0"
cmake --build . --config Release --target ALL_BUILD -- -m
```

change that to 

```
cmake .. -G "Visual Studio 17 2022" -A ARM64 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="%DEPS%/usr/local" -DCMAKE_INSTALL_PREFIX="./OrcaSlicer" -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE="C:/Program Files (x86)/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/VC/VCTargets/Platforms/ARM64/PlatformToolsets/v143/Toolchain.cmake
```

eblieb, 2024-01-01:
Dont think I can since I dont own a copy of Visual Studios...

giveen, 2024-01-01:
> Dont think I can since I dont own a copy of Visual Studios...

Its free, the community edition
https://visualstudio.microsoft.com/vs/

eblieb, 2024-01-01:
I just noticed that.. Installing now and will have to update the build path since that is tied to enterprise, but will give it a try

eblieb, 2024-01-04:
Was unable to build this. The platform tool sets isn't a thing that ever showed up. Not sure if the community version has that ability.

closed 2024-01-14 by eblieb