OrcaSlicer/OrcaSlicer #3262: 1.9.0 Alpha Release Bug on mac.

christianjou, 2023-12-24:
### OrcaSlicer Version

1.9.0 Alpha release

### OS version

OS: MONTEREY 12.7.1

### Additional system information

at start it bug. 
[image]

### Printer

none

### How to reproduce

every launch

### Actual results

Bug

### Expected results

should open

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-12-25:
Will take a look soon.

Noisyfox, 2023-12-26:
Currently you can install zstd as a workaround
`brew install zstd`

We are currently working on a proper fix.

atl285, 2023-12-26:
Just to complete here: https://github.com/SoftFever/OrcaSlicer/issues/3099

SoftFever, 2023-12-26:
> Just to complete here: #3099

Damn, how did I overlooked that

closed 2024-04-02 by github-actions[bot]