OrcaSlicer/OrcaSlicer #3276: Problem with Setting Sparse Infill to Zero

krmtech, 2023-12-25:
### OrcaSlicer Version

1.8.1

### OS version

MacOS 14.2

### Additional system information

2023 MacBook Pro
18GB Unified RAM
Chipset Model:	Apple M3 Pro Type:	GPU Total Number of Cores:	14

### Printer

BambuLab P1P

### How to reproduce

1. Go to "Strength"
2. Set Sparse Infill to "0"
3. Slice
4. Print

### Actual results

When Infill layer prints, "banjo strings" are printed. Horizontal lines of dry filament that are not attached to layer below.
[image]

### Expected results

Either print solid infill
OR
Error out, warning that Infill cannot be set below XX

### Project file & Debug log uploads

[Drawer Organizer.zip](https://github.com/SoftFever/OrcaSlicer/files/13765995/Drawer.Organizer.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-12-26:
This is expected behavior.
0 mean not sparse infill.
If you want solid infill you will need to put 100%

We won't treat 0 as 100% here because there are cases we do want to print with any infill depends on the model shapes.

closed 2023-12-26 by SoftFever

krmtech, 2023-12-26:
Can you please help me understand the extruder behavior when set to 0%?

Is the filament issue as shown in the photo actually a layer above the infill layer(s) and because infill is absent, there is nothing for the filament to adhere to?