OrcaSlicer/OrcaSlicer #3278: I raise two problems

olcayoren, 2023-12-25:
### OrcaSlicer Version

1.9.0 alpha

### OS version

Macos Sonoma 14.0

### Additional system information

Macbook m1 2020 arm

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

I raise two problems.
1- Turkish does not appear in the language list
2- The max xy smooting part under the smooth spiral tab, which opens in spiral vase mode, does not close
I showed the details in the video

https://github.com/SoftFever/OrcaSlicer/assets/33813227/0ad9bdcc-d7c0-445a-97a0-a7679e161b04

### Actual results

None

### Expected results

None

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-12-26:
Thank you!
Should be fixed now

closed 2023-12-26 by SoftFever

olcayoren, 2023-12-26:
Hello, there is an improvement, but there seems to be a logical error. When you turn off the spiral vase mode, shouldn't it disable the smooth spiral?

https://github.com/SoftFever/OrcaSlicer/assets/33813227/8b3b1841-3d77-4cf2-a326-ac1e502abb58

Noisyfox, 2023-12-27:
> Hello, there is an improvement, but there seems to be a logical error. When you turn off the spiral vase mode, shouldn't it disable the smooth spiral?
> 
>  1.mp4

That doesn't really matter though.

commit 2f17c191 (2023-12-26): Fixed a regression that Turkish lang option was missing

Fixed #3278
commit bc76677e (2023-12-30): Fixed a regression that Turkish lang option was missing

Fixed #3278