OrcaSlicer/OrcaSlicer #3279: Hang/Crash when trying to slice with certain support settings

mrelwood, 2023-12-25:
### OrcaSlicer Version

1.8.1

### OS version

macOS Ventura 13.6.1

### Additional system information

MacBook Pro 16" mid 2019

### Printer

Bambu Lab P1P

### How to reproduce

Open the attached .3mf and try to slice.

The object sliced succesfully until I changed some support settings, mainly:
Top interface spacing 
Max bridge length (10 to 2mm)
Tree support branch distance (5 to 2mm)
Tree support branch diameter (5 to 2mm)

The attached project should be the crashing one though.

### Actual results

OrcaSlicer becomes unresponsive. If I click the button to cancel slicing in time, the app will start working again after a very long time (10 to 30 minutes). If I don't, the app will crash after an even longer time. When launched again, OrcaSlicer will suggest loading the project again, which it does succesfully.

### Expected results

For the slicing to complete succesfully.

### Project file & Debug log uploads

[Heinin Ompelupoytatulppa v3 (SLICING HANG).3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13766441/Heinin.Ompelupoytatulppa.v3.SLICING.HANG.3mf.zip)
[Log files.zip](https://github.com/SoftFever/OrcaSlicer/files/13766444/Log.files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

silence48, 2023-12-31:
possible relation to [2921](https://github.com/SoftFever/OrcaSlicer/issues/2921)

mrelwood, 2023-12-31:
> possible relation to [2921](https://github.com/SoftFever/OrcaSlicer/issues/2921), but if it is, then it could help us to figure it out, since all the other ones I saw with similar symptoms are windows builds.

Doesn’t seem similar to me actually. My project had 110 layers, and while it had two modifier blocks, it wasn’t otherwise that complicated. The layers 18 and 19 where the hang happened weren’t special either, no modifiers near those layers, not even support interfaces.

What I did find from further testing was that as I gradually lowered the mentioned support settings, the slicing time did get exponentially longer and longer. I ended up using Bridge length at 3mm and Branch distance & diameter at 4mm to get a reasonable compromise. It hangs at the problem layer for 8 seconds while everything else happens in a blink of an eye.

closed 2024-04-08 by github-actions[bot]