OrcaSlicer/OrcaSlicer #3282: V1.9.0 Alpha Linux crashes when sending file to SD card

emelbardis, 2023-12-25:
### OrcaSlicer Version

V1.9.0 Alpha

### OS version

Fedora Linux / appimage

### Additional system information

11th gen i9, intel graphics

### Printer

Bambu Lab X1C

### How to reproduce

Slice all
Send all

### Actual results

Orcaslicer either crashes and closes or says that I need to be on the same LAN

### Expected results

File sent to X1C SD card

### Project file & Debug log uploads

no logs - will try again if needed

### Checklist of files to include

- [X] Log file
- [X] Project file

MonkaKokosowa, 2023-12-27:
Also crashes while testing lan connection to the printer before sending if the access code and IP are correct

CCS86, 2023-12-28:
Also crashing on Windows x64 when sending to SD card.

CCS86, 2023-12-31:
Still crashing in 1.9.0 beta

CCS86, 2024-01-05:
@SoftFever can you reproduce?

SoftFever, 2024-01-05:
Yes, and I have fixed this issue. Will release 1.9.0 official soon

SoftFever, 2024-01-05:
Fixed.

closed 2024-01-05 by SoftFever