OrcaSlicer/OrcaSlicer #3290: Multiple Extruders, Automatic build plate slicing,

cdvankammen, 2023-12-26:
Is there a reason that there is no is no automatic slicing in this slicing. This is a Prusa Slicer fork. 
Also, a need for multiple extruders is required for this slicer to complete strongly with Prusa Slicer. 

A need or desire to be able to add objects (shapes and text) directly to the build plate as Prusa Slicer has. I would be willing to try helping wit programing with any guidance.

Noisyfox, 2023-12-27:
> Is there a reason that there is no is no automatic slicing in this slicing. This is a Prusa Slicer fork.

Because BBS disabled this for whatever reason, and we haven't brought it back yet.

> Also, a need for multiple extruders is required for this slicer to complete strongly with Prusa Slicer.

That's on the plan.

> A need or desire to be able to add objects (shapes and text) directly to the build plate as Prusa Slicer has. I would be willing to try helping wit programing with any guidance.

It's already supported in the 1.9 alpha version.

cdvankammen, 2023-12-27:
Thank you for the response. With regards to things being disabled and the addition of multiple extruders.

I would like to offer my help with progressing these features. Are you an active developer on the project? Or who could I work with to provide help?

cdvankammen, 2024-01-13:
So I see that the 1.9 was released. However, the options for automatic printing is still not available do you happen to know the pull request that added it?

closed 2024-04-20 by github-actions[bot]

cdvankammen, 2024-04-20:
Any thoughts when this will become a thing?