OrcaSlicer/OrcaSlicer #3292: 1.9.0 alpha Crashes when color painting

Heiliger-Gral, 2023-12-26:
### OrcaSlicer Version

1.9.0 Alpha

### OS version

Windows 10 X64

### Printer

Rook

### How to reproduce

As soon as I want to paint a multicoler print and I click on the "Color Painting" button at the top, OrcaSlicer crashes

### Actual results

[image]

### Expected results

Orca Closes without notification

### Project file & Debug log uploads

[crash_Tue_Dec_26_11_15_11_0.log](https://github.com/SoftFever/OrcaSlicer/files/13770785/crash_Tue_Dec_26_11_15_11_0.log)
[lizard Orca1.zip](https://github.com/SoftFever/OrcaSlicer/files/13770792/lizard.Orca1.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2023-12-26:
Thanks for reporting back.
Does it crash consistently?

Heiliger-Gral, 2023-12-26:
> Thanks for reporting back. Does it crash consistently?

After the 5th time it worked... but I still have a problem in that I get an MCU timer to close error when I print a lot of multicollers...

With Prusaslicer it works without any problems

But it's a different problem and I don't know how best to describe and show it to you... but I wrote about it in the Discord

silence48, 2023-12-31:
Another 
Exception Code :c0000005 ACCESS_VIOLATION

Related issues include #2921 #3375 #3157 #3340

closed 2024-04-08 by github-actions[bot]