OrcaSlicer/OrcaSlicer #3293: OrcaSlicer V1.9.0 Alpha 取消打印并上传后，重新上传会卡死

pudding1111, 2023-12-26:
### OrcaSlicer Version

V1.9.0

### OS version

windows11

### Additional system information

cpu 13600kf
内存 光威6000 16*2
固态 大华c900plus 1t
主板 双敏b760m
显卡 盈通大地之神4060ti

### Printer

启庞KLP1

### How to reproduce

OrcaSlicer V1.9.0 Alpha 发现有一个参数错了，上传并打印  这个进程进行到一半时取消掉，然后重新切片并上传时，打印机会无反应，软件参数界面显示上传进度为0。

### Actual results

OrcaSlicer V1.9.0 Alpha 发现有一个参数错了，上传并打印  这个进程进行到一半时取消掉，然后重新切片并上传时，打印机会无反应，软件参数界面显示上传进度为0。

### Expected results

希望正常

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-12-26:
谢谢反馈
我不能重现这个问题，怀疑是你的打印机固件那里的原因。
看你描述是打印机那里没有响应

sunjinmao, 2024-03-03:
我也是klp1出现这个问题，目前用的1.9.1的版本。以前大概1.8.1那会就没有这个bug

closed 2024-06-09 by github-actions[bot]