OrcaSlicer/OrcaSlicer #3297: Orca 1.9 Alpha no --datadir option anymore

Prutsium, 2023-12-26:
### OrcaSlicer Version

1.9 Alpha

### OS version

Win11

### Printer

Bambu X1C

### How to reproduce

In the past it was possible to point to a specific datadir to be able to have profiles on the network by opening orcaslider with orca-slicer.exe --datadir profiles and it would use the profiles directory within the Orcaslicer directory.

### Actual results

Profiles are not loaded and datadir is not been used

### Expected results

To open profiles from custom location

### Project file & Debug log uploads

..

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-12-27:
Fixed in dev.
Will be available in 1.9.0 beta

Prutsium, 2023-12-27:
Great and thanks.
Btw paypal or something like that? You do a great job and has to be
appreciated 😁

closed 2024-04-03 by github-actions[bot]