OrcaSlicer/OrcaSlicer #3299: Print Sequence By Objetct reports a height collision when collisions is virually impossible

bc30se, 2023-12-26:
### OrcaSlicer Version

1.8,1

### OS version

win11

### Printer

creality k1

### How to reproduce

Try to place two objects of the same height distanced from each other. Then try to slice with PRINT SEQUENCE - BY OBJECT. The Sliver will show the height collision error message even for the arragments that are pysically not possible to collide with the hotend.

### Actual results

You cant slice the object and generate your gcode.

### Expected results

It should be possible to specify maximum radius and allowable height of the object. Or this option should be repaired.

### Project file & Debug log uploads

[crash_Tue_Dec_19_09_31_39_0.log](https://github.com/SoftFever/OrcaSlicer/files/13772947/crash_Tue_Dec_19_09_31_39_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2023-12-27:
What's your clearance between the nozzle tip and the z axis rail?

bc30se, 2023-12-27:
It is about 10cm - Creality K1, however I cannot make new entries as it is possible in Prusa slicer (radius/height)

Noisyfox, 2023-12-28:
10cm? Are you sure? If you are certain then you could update these settings:
[image]

Use at your own risk!

bc30se, 2023-12-28:
Where Can I find these settings, I was looking for them everywhere.

bc30se, 2023-12-28:
Got it man thank you!!

closed 2024-01-04 by Noisyfox

otede, 2024-06-15:
This doesn't make sense. The XY contour should be the only limiting factor for objects of height much less than the Z volume limits. The gantry height should then be used by the slicer to figure out how not to collide with any objects.

otede, 2024-06-15:
The smaller the gantry height value the mor cautious Orca should be, setting higher Z travel up margins. I've just had a 50 mm model collision with mere 2 models on the plate, auto-arranged by Orca! Project attached:
[orca collision very small model pikachu 2 pieces.zip](https://github.com/user-attachments/files/15846387/orca.collision.very.small.model.pikachu.2.pieces.zip)
[image]

otede, 2024-09-22:
This is still happening all the time. Sometimes it resets with the very same objects on the plate by merely re-initializing auto arrange, which proves this is a bug that produces bogus warnings. Reporting again
https://github.com/SoftFever/OrcaSlicer/issues/6876