OrcaSlicer/OrcaSlicer #3300: Orca 1.9.0 alpha - Add new Creality SE, KE profiles don't work as expected

kiswarday, 2023-12-26:
### OrcaSlicer Version

Orca 1.9.0 alpha - Linux appimage

### OS version

Linux Mint

### Additional system information

On the system is already present the 1.8.1 appimage version using the default directories:
config => ~/.config/OrcaSlicer
data => ~/.local/share/orca-slicer
cache => ~/.cache/orca-slicer

### Printer

Creality Ender 3 V3 SE

### How to reproduce

Add the new Creality Ender 3 V3 profile:
1. Click on Select/Remove printers (systems preset)
2. Select Creality Ender 3 V3 SE (or KE) with nozzle 0.4
3. Click confirm

### Actual results

1. Old presets disappear and a new profile, named "Default Printer", appear. The values in the profile are not valid as well (i.e. V3 SE has a bed size of 220 mm but the setting is 200 mm)
2. Closing and reopening the application make reappear again the old profiles plus the new one (with the right values this time)

### Expected results

Add the new selected printer without reopen the appimage

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13773076/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

SoftFever, 2023-12-28:
Yeah, I noticed this too.
Need to further investigate on it.
For now, the workaround works so it's slightly low priority.

bistory, 2024-01-02:
> Yeah, I noticed this too. Need to further investigate on it. For now, the workaround works so it's slightly low priority.

This has been fixed, @kiswarday can you confirm that with one of the latest builds (I don't remember is beta already fixed that issue)

kiswarday, 2024-01-02:
> > Yeah, I noticed this too. Need to further investigate on it. For now, the workaround works so it's slightly low priority.
> 
> This has been fixed, @kiswarday can you confirm that with one of the latest builds (I don't remember is beta already fixed that issue)

Yes, in the Beta version it has been fixed

mkellerman, 2024-01-02:
I have tried the Beta version from 3 days ago, and i can see the Creality Ender-3 V3 KE in the Device tab, but can't seem to print to it.

You mention a workaround, is there a link somewhere?

bistory, 2024-01-03:
> I have tried the Beta version from 3 days ago, and i can see the Creality Ender-3 V3 KE in the Device tab, but can't seem to print to it.
> 
> You mention a workaround, is there a link somewhere?

That's weird, I used it yesterday without any issue, can you try one of the latest nightlies ? This one for example : https://github.com/SoftFever/OrcaSlicer/actions/runs/7385948192

rtorchia, 2024-03-30:
I can confirm I'm having this problem with the latest 2.o release.
Choosing any of the Ender V3 S3 printer's with 0.6 and 0.2 nozzle presents the proper settings:

[image]

But anything for a 0.4 nozzle (even custom presents for the printer) only gives you the Default Settings (and they are the wrong values)

[image]

bistory, 2024-03-30:
@cochcoder there seems to be something wrong with your latest PRs 🙂

cochcoder, 2024-04-01:
> I can confirm I'm having this problem with the latest 2.0 release. Choosing any of the Ender V3 S3 printer's with 0.6 and 0.2 nozzle presents the proper settings:
> 
> ![image](https://private-user-images.githubusercontent.com/20632876/318199994-3bbfb10d-de9b-4118-8e1f-d7f751586728.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5OTAxNTIsIm5iZiI6MTcxMTk4OTg1MiwicGF0aCI6Ii8yMDYzMjg3Ni8zMTgxOTk5OTQtM2JiZmIxMGQtZGU5Yi00MTE4LThlMWYtZDdmNzUxNTg2NzI4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDE2NDQxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlNGVjOWU4OGI4MTBiYzZlZTkxNDIxMDZmY2FkZWQ0YThiZWJlYTgxYTE4NTdjNmM1ZTlkNjQyODU5OWVmNzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.T33UGCm2DmJmk9sIWJLIeMzo0AyxCuC6MYKJWEWJcOM)
> 
> But anything for a 0.4 nozzle (even custom presents for the printer) only gives you the Default Settings (and they are the wrong values)
> 
> ![image](https://private-user-images.githubusercontent.com/20632876/318200132-39c0917a-d399-44bc-99d8-f5c48d7a7ceb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5OTAxNTIsIm5iZiI6MTcxMTk4OTg1MiwicGF0aCI6Ii8yMDYzMjg3Ni8zMTgyMDAxMzItMzljMDkxN2EtZDM5OS00NGJjLTk5ZDgtZjVjNDhkN2E3Y2ViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDE2NDQxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcxOWRiM2EwNWUzMzU3NDhmNzliM2EzMzg0NTE0NWIzMTUwYjYzMTc4Y2ViMTdmNDYyMzk1MWIwODc2Nzk2MWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8K9WVXmCD3X6zsN7_LVQmYIq6B_nYwP1NwiCNEgLIaM)

This should be fixed by #4785

rtorchia, 2024-04-01:
I manually edited the required files to get everything back to normal. I'm glad I was able to retrieve everything, but let's not do this again! 😉

closed 2024-07-09 by github-actions[bot]