OrcaSlicer/OrcaSlicer #3303: Uniform filling

sssvolk, 2023-12-27:
When filling the top layers, if the fill type is monotonous, and if there are objects above, prints the fill unevenly.

It is expected that when filling the top layers, if the monotone fill type is selected, and if there are objects above, it will print the entire fill evenly, monotonically.

p.s. now its best sliser!
[image]
[image]

Noisyfox, 2023-12-27:
Try set "only one wall on top threshold" to 0%.

sssvolk, 2023-12-27:
> Попробуйте установить "только одну стену на верхнем пороге" до 0%.

It has become much better, although there are still artifacts. =)
[image]

closed 2024-04-03 by github-actions[bot]