OrcaSlicer/OrcaSlicer #3308: Top infill connection to perimeters setting

m-klecka, 2023-12-27:
**Is your feature request related to a problem? Please describe.**
Not really but could improve quality of prints

**Which printers will be beneficial to this feature**
Probably every printer

**Describe the solution you'd like**
Implement this feature from superslicer https://ellis3dp.com/Print-Tuning-Guide/articles/troubleshooting/small_infill_areas_overextruded.html#not-connected-top-infill-superslicer

**Describe alternatives you've considered**
None

**Additional context**
More info in link

igiannakas, 2023-12-27:
Isn't this the same as monotonic lines top infill that is already present?

m-klecka, 2023-12-27:
> Isn't this the same as monotonic lines top infill that is already present?

You're right, i missed that:) Marking as closed

closed 2023-12-27 by m-klecka