OrcaSlicer/OrcaSlicer #3310: Add the calibration cube

mirspan, 2023-12-27:
Add the calibration cube to the test in the slicer. I have prepared a calibration cube and attached it to the message in the archive *.zip.

[calibration-cube.zip](https://github.com/SoftFever/OrcaSlicer/files/13778287/calibration-cube.zip)

giveen, 2023-12-27:
It actually already has several calibration prints

[image]

mirspan, 2023-12-27:
Why are the calibration cube and Autodesk FDM Test missing from the top calibration menu?

[image]

Eldenroot, 2023-12-27:
Because calibration cube could be set how you need (print settings). This is not a bug.

closed 2024-04-03 by github-actions[bot]