OrcaSlicer/OrcaSlicer #3311: Filament Load / Unload time didn't change ETA

Gnierat, 2023-12-27:
### OrcaSlicer Version

1.8.1 and 1.9.0

### OS version

Windows 11

### Printer

Voron 2.4

### How to reproduce

Slice
change load / unload time
Slice
Same time

### Actual results

Didn't change ETA

### Expected results

Recalculate ETA

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13778594/log.zip)
[oddish.zip](https://github.com/SoftFever/OrcaSlicer/files/13778596/oddish.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

eoyilmaz, 2024-01-15:
Yes, exactly what I was going to report, here are some screen shots:

Before
-------
Filament Load Time: 45 s
Filament Unload Time: 25 s
Total Time: 3h 48m

After
-------
Filament Load Time: 450 s
Filament Unload Time: 250 s
Total Time: 3h 48m

[image]
[image]

eoyilmaz, 2024-01-16:
Started looking into this 👍

eoyilmaz, 2024-01-16:
There is an issue the `machine_filament_load_time` / `machine_filament_unload_time` which stores the printer's filament load/unload time is never visible in the UI, and as far as I see the per filament `filament_load_time` is not read at all.
 
[image]

eoyilmaz, 2024-01-16:
Updating the `machine_load_filament_time` and `machine_unload_filament_time` in the `.json` file of the printer now calculates the print time correctly:

[image]
[image]

eoyilmaz, 2024-01-16:
So the `machine_load_filament_time` and `machine_unload_filament_time` are disabled on purpose here:
https://github.com/SoftFever/OrcaSlicer/blob/main/src/slic3r/GUI/Tab.cpp#L4059

and instead of `filament_load_time` and `filament_unload_time`, the hidden and 0 by default `machine_load_filament_time` and `machine_unload_filament_time` are used here:
https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/GCode/GCodeProcessor.cpp#L1039

I think replacing those with the `filament_load_time`/`filament_unload_time` should fix this... Let me try it quickly.

eoyilmaz, 2024-01-16:
It seems that Bambulab did change the working code of PrusaSlicer, I'm migrating this portion of the code from PrusaSlicer again.

eoyilmaz, 2024-01-16:
Yep... it seems to be working now 👍 

creating PR in a minute.

eoyilmaz, 2024-01-16:
PR created 👍

commit 25278841 (2024-01-25): Fix #3311: filament Load / Unload time was not used in time estimation (#3706)

* Fix for #3311:

The `config.filament_load_time.values` and `config.filament_unload_time.values` are completely ignored. This was working in PrusaSlicer and apparently BBS converted the `GCodeProcessor::TimeProcessor.filament_load_times` and `GCodeProcessor::TimeProcessor.filament_unload_times` from being `std::vector<float>` to just `float` and they were using the BBS specific and currently hidden `machine_load_filament_time` and `machine_unload_filament_time` config values. Reverted that change by copying those  [... cut]
commit a16a1e93 (2024-08-11): Feature/mmu toolchange time (#6426)

* Revert "Fix #3311: filament Load / Unload time was not used in time estimation (#3706)"

This reverts commit 252788419bbf3ab127dfd5ca869497e716f11774.

* revamp tool/filament change time

* tweak message
closed 2024-01-25 by SoftFever