OrcaSlicer/OrcaSlicer #3313: Perimeters offset too much for one wall on top surfaces

vovodroid, 2023-12-27:
For lower top surfaces perimeters are shifted inward, especially when there are number of walls:

[image]

I guess instead of complex calculations here
https://github.com/SoftFever/OrcaSlicer/blob/7ab2fabe6a8b314facb05a0cea372f5b304dc63a/src/libslic3r/PerimeterGenerator.cpp#L885

something like `coord_t offset_top_surface = ext_perimeter_width;` could produce better result:

[image]

providing base for next layer external perimeter. Picture is from Prusa with ported code.

[Orca-Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/13778968/Orca-Cube.zip)

Eldenroot, 2023-12-29:
Are you working on a PR?

vovodroid, 2023-12-29:
Not really. Merill took this code to his SS and used some other formula - https://github.com/supermerill/SuperSlicer/issues/4022#issuecomment-1871368000

PR #6236 (2024-09-23): Arachne "one top wall" fix and refactor
This PR supersedes  _Reduce one top perimeter offset #6086_ and fixes  #6000 _One wall on top causes wrong wall generation in Arachne_ and  fixes #3313 _Perimeters offset too much for one wall on top surfaces_.

See previous discussion in **Reduce one top perimeter offset #6086**.

Changes:

1) Arachne one top level wall code was replaced by ported from Prusa, while keeping "One wall threshold" parameter.
2) Classic - perimeter offset was reduced to produce good result.

@igiannakas @Eldenroot FYI
closed 2024-09-23 by SoftFever