OrcaSlicer/OrcaSlicer #3320: Elegoo thumbnail seems not working in 1.90a

sufe08, 2023-12-28:
### OrcaSlicer Version

1.90 alpha

### OS version

win10

### Printer

elegoo neptune 4

### How to reproduce

when sliced a model with printer neptun 4 in orca 1.90a where thumbnail format set as default "ColPic", the output gcode gives the same thumbnail block as set format"PNG", so that the preview will not work on Neptune4 screen

### Actual results

the output gcode gives the same thumbnail block as set format"PNG", so that the preview will not work on Neptune4 screen

### Expected results

The output gcode should give the thumbnail block as ";Simage ..... ;;Gimage....."

### Project file & Debug log uploads

[left-bottom.zip](https://github.com/SoftFever/OrcaSlicer/files/13782469/left-bottom.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

sufe08, 2024-01-04:
I found "Gimage""Simage" section at the final part of output  Gcode file and seems useless for Elegoo screen to show preview

Barsik95, 2024-01-18:
haben sie mittlerweile eine Lösung für das Problem gefunden?

StealthES, 2024-01-30:
wann ist eine Lösung in Sicht? Gibt es eine Möglichkeit über einen externen Prozess?

troygeiger, 2024-02-23:
I was seeing the same issue with thumbnails and the metadata as well. It looks like Elegoo customized the moonraker metadata.py to parse the simage and gimage data but only implemented the methods for Cura. That causes an error when parsing gcode generated by any slicer other than Cura.  The metadata.py file on moonraker's repo doesn't parse simage & gimage data from what I can tell so this seems to be Elegoo specific. 
All that being said, I doesn't look to be an issue caused by OrcaSlicer.
I posted my findings on their reddit so hopefully that will fix it and put our a firmware update.