OrcaSlicer/OrcaSlicer #3323: Crashes immediately after startup on ArchLinux (appimage)

shebang79, 2023-12-28:
### OrcaSlicer Version

1.9.0-alpha

### OS version

Archlinux

### Additional system information

5800x3d, 32 gigs ram, 4090

### Printer

None

### How to reproduce

Run appimage, gui loads up then immediately crashes

### Actual results

gui loads up then immediately segfaults

### Expected results

Program doesn't segfault

I had this issue on Bambu Studio caused by a recent update to the Bambu network plugin, I rolled back to a previous version and it works fine. I tried the same fix here and it also worked. It asks to update the network plugin and it continues to work so long as you don't update.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13784623/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

etrigan63, 2023-12-29:
Can confirm this behavior. ArcoLinux (Arch) + Hyprland (Wayland). Ryzen 9 5950X, 64GB RAM, 6750XT

WorldTeacher, 2024-01-01:
Happens to me as well, both on 1.8.1 and 1.9.0 alpha
appimage startup in console looks like this:
https://pastebin.com/iCKFccjE

The same happens with the AUR package (https://aur.archlinux.org/packages/orca-slicer-bin). On my system, the issue seems to be related to a package upgrade, as everything was fine yesterday before I updated these packages:
- upgraded python-regex (2023.10.3-1 -> 2023.12.25-1)
- upgraded python-pytest (7.4.3-1 -> 7.4.4-1)
- upgraded lib32-mesa (1:23.3.1-1 -> 1:23.3.2-1)
- upgraded iputils (20231222-1 -> 20231222-2)
- upgraded icoextract (0.1.4-2 -> 0.1.4-3)
-upgraded firefox-developer-edition (122.0b3-1 -> 122.0b4-1)
- upgraded mesa (1:23.3.1-1 -> 1:23.3.2-1)
- upgraded fontconfig (2:2.15.0-1 -> 2:2.15.0-2)

System: XeroLinux (Arch) + KDE (X11) Ryzen 7 5800X, 64 GB Ram, RTX 3090

shebang79, 2024-01-01:
Yup, same here. Some very recent update broke all all versions of Orca (including the latest beta) as well as Bambu Studio. I'm at work but taking a guess I'd start with rolling back Mesa.

WorldTeacher, 2024-01-01:
> Yup, same here. Some very recent update broke all all versions of Orca (including the latest beta) as well as Bambu Studio. I'm at work but taking a guess I'd start with rolling back Mesa.
> 
> If that's not it, maybe fontconfig or Python?

rolling back mesa to 1.23.3.1-1 enables me to launch 1.8.1 and 1.9.0

closed 2024-04-09 by github-actions[bot]