OrcaSlicer/OrcaSlicer #3324: inner-outer-inner/infill option move wipe to inner layer instead of outer.

hannoHI, 2023-12-28:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Printer

Voron 2.4 / Prusa Mk3S

### How to reproduce

1. Goto Quality, Advanced, Order of inner wall/...
2. Select inner-outer-inner/infill
3. must have 3 wall loops
4. wiping displayed on Line type

### Actual results

Wiping moves from outer layer to a inner layer. This removes the benefit of wiping from this sequence.
[image]

### Expected results

Expect the external wiping to stay on the external wall.
[image]

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/SoftFever/OrcaSlicer/files/13785010/Cylinder.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2023-12-28:
It's because there is no retraction when printing inner outer inner mode. so there is nothing to wipe. We could look at enforcing a wipe operation always even if there is no retraction instructed but not sure if this would benefit or hurt print quality.

You can try that by setting your travel distance threshold to 0, it will then always retract and wipe, including the external perimeters.

igiannakas, 2023-12-28:
It would be great if you could do some print tests with the above option enabled (travel distance to 0) and report back with before and after seam shots. This would help evaluate whether this option would offer a benefit.

closed 2024-04-04 by github-actions[bot]

Rickthebig, 2024-07-19:
> It would be great if you could do some print tests with the above option enabled (travel distance to 0) and report back with before and after seam shots.

I came accross this question since I am in a battle against stringing using PETG. I can readily print usual things with acceptable quality, but for the sake of it, I often try challenging models, which are not expected to go well with this material.

I decided to try to print this model in PETG just impeccably, without any signs of stringing, both using Orca and Cura. This is just an experiment, but one I wanted to get straight. I even changed my printer setup from Bowden to Direct Drive to not have an excuse:

[image]

After many iterations and several days improving the outcome, this is the final result (Orca on the left with 3 mm retraction, Cura on the right with 1 mm retraction):

[image]

Yes this is quite difficult using PETG (towers are a few mm apart), and there is a school of 3d printer enthusiasts who profess that stringing is unavoidable using PETG.

I won't go through all things I tried but would rather come to the point, why Cura turned out to be much better in this case.
As you know, following factors are crucial when fighting stringing (let's not mention the obvious - temperature and retraction length):

1. how fast are the travel moves between retraction/deretraction points
2. how clean is the nozzle after being wiped right before the travel move
3. whether the nozzle will overfly the next border at the target spot without leaving accumulating edges (which grow slopingly if the travel moves are repetitive).
4. how strong is the pressure just at the begin of the travel move (!)

Both Cura and Orca can be equally fast while travelling (of course), but Orca is able to wipe the nozzle better, even combining the wipe move with retraction. Point 3 goes also to Orca, since it offers different ways of performing a z hop, although none of them is seems to be optimal.

So why is Cura better in printing this piece? Because it allows to *depressure* the filament by using coasting. Look at this:

[image]

You can see that the nozzle moves further along the extrusion path without the stepper pushing material. In the case of compressible PETG this is a very appropriate way to relieve pressure by letting the filament just flow.

Now, Orca has no coasting but you are proposing an alternative method instead:

>"You can try that by setting your travel distance threshold to 0, it will then always retract and wipe, including the external perimeters."

>"Set retract before wipe to 0 if you want to use the "ooze" to close the gap."

Let's take a look if the gap can be closed as in Cura:
I defined a "seam gap" of 3 mm (equivalent to defining a coasting volume in Cura), set a "travel distance threshold" to 0, and "set retract before wipe" to 0.

[image]

[image]

Here is the result:

[image]

As you can see, the extrusion path changes. The nozzle does not coast along the original path oozing out material without stepper help and depressurizing the filament. 

This is what the poster @hannoHI here requested
https://github.com/SoftFever/OrcaSlicer/issues/3325#top

So it just does not work as you apparently think. With your method it is not possible to emulate the coasting function in Cura.

Which would be very desirable. But developers think they always know better, so I am afraid that will never happen. I will be forced to use Cura when using PETG. Forever.

Late post, but still relevant I'm afraid.