OrcaSlicer/OrcaSlicer #3328: Rename Exlude Objects

ProtekDev, 2023-12-28:
**Is your feature request related to a problem? Please describe.**
When then .stl Filename contains special Characters, the print will never start and i get no error on Klipper.
Only the .gcode Filename will get fixed atm, but not the exlude objects names.

**Which printers will be beneficial to this feature**
Every Klipper Printer

**Describe the solution you'd like**
Like the already implemented File rename when uploading / printing directly with Orca.
Apply this same to the exlude objects.

**Describe alternatives you've considered**
A general rename of all files like:
Datetime_Part.gcode + exclude objects

SoftFever, 2023-12-29:
can you provide a sample project?
I added a feature to replace some special characters a while ago.
If you can provide a sample project, I can take a look

ProtekDev, 2023-12-29:
Sure, that what i had.
I was just wondering if my printer config is broken because the printer never started to heat up.
The .stl has my original naming with special characters and the .gcode seems to got fixed by the orcaslicer, like described above.

Klipper must have stucked at Line 280 (but no error):
EXCLUDE_OBJECT_DEFINE NAME=TPU_Fuß.stl ....

[Sample_SpecialChars.zip](https://github.com/SoftFever/OrcaSlicer/files/13795757/Sample_SpecialChars.zip)

SoftFever, 2023-12-30:
That looks like a legitimate Unicode character. Make sure your klipper support Unicode or use ascii only in your model names

ProtekDev, 2023-12-30:
How can i do that ?
As i figured out, it should be a problem of the Python version.
For my InputShaper i have to use Python2, so im kinda limited.
Sure i can try to correctly name my .stl Files in the Future.
Or create a auto rename script by myself, before slicing.

SoftFever, 2023-12-30:
Upgrade to python 3 is the easiest way I'd say.
And it doesn't affect input shaper unless you are referring to some self made component which only supports python 2?

closed 2024-01-05 by SoftFever