OrcaSlicer/OrcaSlicer #3329: SVG files loading incorrectly

JesseHavok, 2023-12-28:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Additional system information

Intel Core i7-10750
32gb ram
NVIDEA RTX 2070 Super

### Printer

Qiditech X-Max-3

### How to reproduce

Create vector in Illustrator
Load or drag svg file into OrcaSlicer

### Actual results

SVG files load into the program with manifold errors as if some anker points are missing. 
[image]

### Expected results

Design should be complete

### Project file & Debug log uploads

[bad SVG.zip](https://github.com/SoftFever/OrcaSlicer/files/13786026/bad.SVG.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-12-28:
Have you tried the 1.9?

JesseHavok, 2023-12-29:
> Have you tried the 1.9?

Yes

Noisyfox, 2024-01-03:
Could you upload the svg file?

JesseHavok, 2024-01-03:
[image]

Noisyfox, 2024-01-04:
[image]
It works alright on 1.9 beta.

closed 2024-04-12 by github-actions[bot]