OrcaSlicer/OrcaSlicer #3330: Feature request for fuzzy paint or to prevent fuzzy skin leaking to internal surfaces

nickion, 2023-12-28:
Fuzzy skin is a useful feature, however it leaks through holes. In my case I have a box with a hole for a USB socket in the side and openings in some other places, and fuzzy skin then wraps around the contour to the inside. Using a modifier is problematic because of the issue with fuzzy skin being applied to the boundary between modifiers and the model.  I wonder whether adding a feature to block holes for the fuzzy algorithm would be a viable approach to solving, or perhaps a fuzzy painter where fuzzy skin can be added and removed by clicking on faces. The painting idea could be more general and useful overall.

t-ihmels, 2024-01-18:
My own scenario is a box with a lid.  Where the box steps in for the lid I don't want fuzzy, but the box walls below I do.

nickion, 2024-04-18:
Not sure whether this is a bug or best as an enhancement, but let's get it tagged as either one of those and move it along.

ph0tic, 2024-07-21:
pls do this

TylonHH, 2024-08-18:
It's not a feature. Its relatetd to: https://github.com/SoftFever/OrcaSlicer/issues/5185
[image]

ProfessorKaos, 2024-09-18:
THis really would be quite a game-changer for home-manafacturing. I have huge respect for the competent devs working Orca, and am very grateful. This would definitetly be a huge upgrade for my process and work. I really wish I had the skills to contribute!

TylonHH, 2024-12-18:
As a workaround there is 3rd party solution: https://youtu.be/cNkHfydnUCI

tlhintoq, 2025-02-02:
Fuzzy skin texture painting tool is clearly a desired item.  Just sayin'

https://github.com/SoftFever/OrcaSlicer/issues/7936
https://github.com/SoftFever/OrcaSlicer/issues/4471
https://github.com/SoftFever/OrcaSlicer/issues/3330

tlhintoq, 2025-05-04:
Not stale

jonathanstarke, 2025-05-05:
Please please, I need this feature...