OrcaSlicer/OrcaSlicer #3332: Kobra 2 Series Update? Neo/Plus/Max

crazybanane, 2023-12-28:
Hello will there comming soon a update on the Anycubic Kobra 2 Series? e.g Plus Max Neo?

csstup, 2023-12-28:
Following.   I'm happy to provide a machine config for the Kobra 2 Pro and help test.    

Attached is the ini for the Kobra 2 Pro from the manufacturer supplied with the machine.  

[Kobra 2 Pro V1.0.ini.zip](https://github.com/SoftFever/OrcaSlicer/files/13789876/Kobra.2.Pro.V1.0.ini.zip)

n4tur3502, 2024-01-10:
I would love to see the kobra 2 plus added.

unbobby, 2024-02-02:
[Anycubic Kobra 2 Max 0.4 PETG自建.zip](https://github.com/SoftFever/OrcaSlicer/files/14141738/Anycubic.Kobra.2.Max.0.4.PETG.zip)
I have created a preset for PETG on the Kobra 2 Max. Any feedback is welcome.

N0rdskog, 2024-03-23:
I started tinkering with porting some of AnyCubic's native settings to a Kobra 2 Pro profile. Does anyone know whether .ini files can be used in any way within OrcaSlicer?

PR #7082 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3
Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes  #3332, closes #6200, closes #5048, and closes #6269
commit 09afb390 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3 (#7082)

Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes #3332, closes #6200, closes #5048, and closes #6269
closed 2024-06-30 by github-actions[bot]