OrcaSlicer/OrcaSlicer #3333: do not log in

rafajluque, 2023-12-28:
### OrcaSlicer Version

1.8.1

### OS version

Windows 10

### Printer

Ender3 S1

### How to reproduce

home, enter/registration

### Actual results

In Windows 10, it won't let me log in, the window remains blank:

[image]

### Expected results

A window should appear to log in to your bambulab account.

### Project file & Debug log uploads

not relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

silence48, 2023-12-31:
You likely need to install the edge webview runtime.

 - [MicrosoftEdgeWebView2RuntimeInstallerX64](https://github.com/SoftFever/BambuStudio-SoftFever/releases/download/v1.0.10-sf2/MicrosoftEdgeWebView2RuntimeInstallerX64.exe)
          - [Details of this runtime](https://aka.ms/webview2)

rafajluque, 2023-12-31:
Perfect, thank you very much, solved with #2004, deleting folders and restarting. Thank you so much

closed 2024-01-01 by rafajluque