OrcaSlicer/OrcaSlicer #3337: overhang detection

Jaze20, 2023-12-29:
### OrcaSlicer Version

1.8.0-rc2

### OS version

win10

### Printer

wanhao duplicator 6

### How to reproduce

30% overhang, 3 wall, 0.48 inner wall, 0.4 outer wall.  I don't know if Orca does it "right' and Prusa was wrong.  But the fact of the matter is that the prusa gcode prints higher quality and doesn't unncessarily slow down

### Actual results

[image]

### Expected results

[image]
the green band is from a modifier.  I know the speed legend is different but the point is to show the overhang detection

### Project file & Debug log uploads

[Overhang detection.zip](https://github.com/SoftFever/OrcaSlicer/files/13790479/Overhang.detection.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2023-12-29:
You have set your overhang speeds backwards. It should be 100%-100%-12-13 not 12-12-100%-100% :)

Jaze20, 2023-12-29:
I just tried reversing it. No overhangs were detected at that point.  It'd be more obvious if there was a hole inside the hex shape.

Jaze20, 2023-12-29:
[Overhang detection.zip](https://github.com/SoftFever/OrcaSlicer/files/13796079/Overhang.detection.zip)

I've attached a new project file with a hole at the center.

Jaze20, 2024-01-04:
Here's another example where a vertical wall is detected as overhang and the fan is turned out as shown by red lines and overhang print speed is used.  If this was PLA, I'm sure this only marginally slows down the print.  But for materials like PETG, this unnecssary use of fan will cause weak brittle parts.  Thanks
[image]

Jaze20, 2024-01-04:
> You have set your overhang speeds backwards. It should be 100%-100%-12-13 not 12-12-100%-100% :)

okay okay I see what you mean now.  Is the fan speed for overhang working as intended?

Jaze20, 2024-01-05:
I'm closing this issue.  The alternating layers where the fan turns on a wall with near zero overhang was due to "cooling fan start up time"  I had it at 0.5s.  I guess that long length took 0.5s to travel.  Still don't feel good I can't lessen the fan speed for less overhang.  I'll see if I can compensate for it with faster print speeds for various overhangs.

closed 2024-01-05 by Jaze20