OrcaSlicer/OrcaSlicer #3339: Preview problems with black filaments

TinyMito, 2023-12-29:
### OrcaSlicer Version

1.9.0

### OS version

macOS 14

### Printer

Bambu Lab A1 Mini

### How to reproduce

See known bug from Bambu Studio here: https://github.com/bambulab/BambuStudio/issues/2761

1. Just load up any PETG / PLA black colour filaments. The models will be pitch black in the preview.

### Actual results

Result:

[image]

### Expected results

Expected:

[image]

### Project file & Debug log uploads

3MF file included for sample.

[Toronto.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13791265/Toronto.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

foreachthing, 2023-12-29:
Easy fix: don't use black. 😉 
Otherwise you'd have to use things like: raytracing, subsurface scattering, shine, roughness, etc.

TinyMito, 2024-03-23:
Confirmed and closing this bug report. Looks like it is fixed.
[image]

closed 2024-03-23 by TinyMito