OrcaSlicer/OrcaSlicer #3340: Crash while slicing with 1.9.0-alpha

kingstar42, 2023-12-29:
### OrcaSlicer Version

1.9.0-alpha

### OS version

Windows 10 64 bit

### Additional system information

CPU: Intel Core i5-7400T 
Memory: 8 GB 
Graphics: Intel HD Graphics 630

### Printer

Voronized/Klipperized Ender 3V2

### How to reproduce

Load project file and slice plate. 

### Actual results

System will either crash and close window, or will show a fatal error and prompt to save and restart. File slices fine in 1.8.1.

### Expected results

Slicing completes successfully.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13791414/log.zip)
[Crash.zip](https://github.com/SoftFever/OrcaSlicer/files/13791416/Crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

silence48, 2023-12-31:
This issue also is also c0000005 ACCESS_VIOLATION
Many related issues have been reported, we really should find one issue where people can comment on this with their logs, instead of having tens of the same thing (IMO)
Related issues include #2921 #3375 #3157

SoftFever, 2024-01-01:
It's a regression bug cause by PR #3014

closed 2024-01-01 by SoftFever

SoftFever, 2024-01-01:
Hi 
Can you try this build? https://github.com/SoftFever/OrcaSlicer/actions/runs/7377220740
It should be fixed now.

commit a36df660 (2024-01-01): Revert "Improve ExtrusionLine::simplify, eliminating many very-short extrusion segments which led to blemishes in thin-wall models sliced with Arachne (#3014)"

This reverts commit 3ca290d503d3eb2c57a6a39d50f7cd5f6a0ea072.

Fixed #3340 #3375
commit 0a223c5f (2024-01-01): Revert "Improve ExtrusionLine::simplify, eliminating many very-short extrusion segments which led to blemishes in thin-wall models sliced with Arachne (#3014)"

This reverts commit 3ca290d503d3eb2c57a6a39d50f7cd5f6a0ea072.

Fixed #3340 #3375

(cherry picked from commit a36df66083ec51627e36dee7e06070156f372ff5)