OrcaSlicer/OrcaSlicer #3342: Printer profile for the Creality CR-M4

rsvmakkie, 2023-12-29:
**Is your feature request related to a problem? Please describe.**
No

**Which printers will be beneficial to this feature**
Creality CR-M4

**Describe the solution you'd like**
Please provide a printer profile for the Creality CR-M4

sinrider, 2024-03-13:
Yes, please.

apache07x, 2024-03-15:
Here's what I've been working on but it's for use with a Sonic Pad so may not be suitable for everyone. I actually haven't used this exact version yet. Similar to what I got printing now but not with the speeds turned up this far. I will be using it on my next print job though
[Creality CR-M4 0.6 nozzle-sonic_pad-fast.zip](https://github.com/SoftFever/OrcaSlicer/files/14610517/Creality.CR-M4.0.6.nozzle-sonic_pad-fast.zip)

Octipole, 2024-03-18:
Thanks @apache07x New to 3D printing. I have a CR-M4 with a Sonic Pad. Since I changed to a 1mm nozzle I'm getting next to no adhesion. Your profile was the best I got so far but I still got a way to go to get back to the occasional Y shift with the 0.4mm nozzle and stock filament. I updated to the Sonic Pad on day one. I recently changed to a cinder block base and liquid cooling. I only got as far as PID calibration. Can anyone provide a profile for the CR-M4 with a Sonic Pad and a 1mm nozzle? Thanks for any help.

apache07x, 2024-03-19:
@Octipole you can change your nozzle size in Orca to 0.1 and calibrate your filament for that size nozzle. This profile should still work you just have to tune your filament to it. Make sure to change your nozzle size in your printer.cfg file as well it's under the extruder section. At a minimum I do flow calibration and Pressure Advance on all of my filament profiles. If you do this make sure pressure advance is 0 on your Sonic Pad so Orca and the Sonic Pad aren't fighting each other. I'll try to upload more filament profiles as I create them. I just became a Hatchbox Distributor for my store so I'll be super busy doing profiles on all 7 of my printers.

Octipole, 2024-04-28:
Thanks for the tips. The combination of those as well as stepping down to 0.6mm nozzle seem to be working until I started getting random disconnects and went back to trying a 1mm nozzle. Subpar usb ports on the Sonic pad might be the problem. Relative to other Creality bedslingers info about CR-M4 still seems to be lacking. I would like to take the plunge into replacing the control board all together but much like profiles info about config files is missing. Thanks for all the help. 

+1 
Describe the solution you'd like
Please provide a printer profile for the Creality CR-M4 (1mm nozzle would be great the machine is very large)

closed 2024-08-05 by github-actions[bot]