OrcaSlicer/OrcaSlicer #3349: Non-manifold edges in calibrations

LJKraus, 2023-12-29:
### OrcaSlicer Version

1.9.0-alpha  and previous versions

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Run OrcaSlicer calibration for a temperature tower or maximum flow rate.

### Actual results

[image]

[image]

Both calibrations use models with non-manifold edges.  This is a minor annoyance in Windows, requiring an extra click to repair.  I'm not sure how MAcOS or Linux handle this without Windows' repair capability.

### Expected results

Replace the test models with versions that have no non-manifold edges.

### Project file & Debug log uploads

[Max volumetric speed test.zip](https://github.com/SoftFever/OrcaSlicer/files/13794031/Max.volumetric.speed.test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

X1phos, 2023-12-29:
I have the same issue on mac for retraction test, VFA test, and max volumetric flow rate test. Unable to repair.

luco85, 2023-12-30:
i have this error on 1.8.1 with Mac but only with retraction test

luco85, 2023-12-30:
i have this error on 1.8.1 with Mac but only with retraction test
[image]

alxgarza, 2024-01-03:
The error is still present on V 1.9.0-beta
[image]

bpjohannessen, 2024-04-23:
+1 For temperature tower with "14 non-manifold edges".

platanax, 2024-05-09:
I have the issue on retractation with version 2.0.0 on Mac

sergtimosh, 2024-05-12:
[image]
Still an issue
Mac- 14.3 (23D56) ARM
Orca - 2.0.0

timnolte, 2024-05-23:
I have these issues on Linux as well. I don't have a Windows machine readily available to repair the models.

jimmystoba, 2024-05-28:
I've just come across the same problem.  However, problem resolves when switching to inner/outer walls from inner/outer/inner.
Hope this helps.

timnolte, 2024-05-30:
> I've just come across the same problem. However, problem resolves when switching to inner/outer walls from inner/outer/inner. Hope this helps.

Unfortunately for me I was already using inner/outer.

I managed to get a Windows 10 install setup in a VM on a different Linux machine and I still get this error, but I can repair them on Windows. This means that I have to use this Windows VM to perform filament calibrations and then export to my main Linux machine, which is a pain.

Zsmith32, 2024-06-11:
I'm experiencing this as well. Was on 1.9.1, tried upgrading to 2.0.0 same issue. 

windows 10

sovol sv06plus

Adrou, 2024-08-04:
[image]

Same here... MacOS.

Ergonomicmike, 2024-08-10:
I don't know if this is related. But today I just noticed that the ORCA lettering doesn't appear on all the sections of the temp tower.

[image]

kinghat, 2024-08-31:
> I've just come across the same problem. However, problem resolves when switching to inner/outer walls from inner/outer/inner. Hope this helps.

this seems to be happening to all my models when using inner/outer/inner on linux appimage 2.1.1.

nickboul, 2024-09-05:
The same here. Do we select repair or leave it?

kinghat, 2024-09-05:
> The same here. Do we select repair or leave it?

i dont think theres actually an issue with the models. also, cant be "repaired" on linux.

hotellonely, 2024-10-02:
It's still happening on 2.2 beta2, also on Mac ARM64.

MermaidMel, 2024-10-07:
> I've just come across the same problem. However, problem resolves when switching to inner/outer walls from inner/outer/inner. Hope this helps.

I changed mine to Outer Inner and it worked when sliced

blipside, 2024-11-23:
I just downloaded Orca Slicer to try it out, v2.2.0, and I am having this problem for the Temperature Tower, Retraction Tower, and Speed Test Structure.  The Flow Rate worked fine, The Tolerance Test and Pressure Advance do not give the error.  I am using Linux, so I cannot repair the models.
OS:  Linux Mint 21.3 (Cinnamon 6.0.4)
Kernel:  5.15.0-126-generic

melbiz447, 2024-12-10:
I have the same problem here with temp tower, retraction test, max flow rate test. Orca 2.2, on Mac 14.7.1, Intel processor. Is there a way to export the file, correct it and import it back with the temperatures settings?

Nam-Ram, 2025-01-17:
Same error on the temperature tower, Linux Mint. Orca 2.2.0  It did let me slice and print though.

blipside, 2025-01-17:
Nam-Ram,
Did it print alright?

Nam-Ram, 2025-01-18:
It printed...but there was no variation between different levels (stock 190-230). I've run temp towers in Cura and there are usually some that fail, none of these did. I checked the g-code and it seems the nozzle temps were correctly set to change per level. Not sure if the V3 KE with this filament is incredibly forgiving, has to do with the manifold, or something else - this is a new printer to me as well as OrcaSlicer, still learning.

I'll run the retraction test soon and report back on that.

thkrmr, 2025-02-14:
Any workaround for this besides Windows? The issue still persists.

Nam-Ram, 2025-02-14:
Haven't gotten a chance yet, going to try to make them manually to avoid the manifold issue.

davidAlittle, 2025-02-17:
Just a fun observation: you can see in the official wiki, they also see the error, but ignore it: 

[image]

thkrmr, 2025-02-18:
I found no other way but to run a Win11 ARM64 VM in UTM App, Orca inside it and apply the fixes. It's annoying, second set of profiles, VM overhead etc. 3rd party temp towers for example cause the same error, as the orca temp tower etc. So an actual Windows machine or VM seem to be sadly the only option at the moment.

tynielsen, 2025-04-21:
I'm betting they're aware of it. Either they don't know how to fix it, or don't care to. I've seen this error popup a handful of times and everytime I ignore it, Slice Plate, and Print (and I have yet to see that warning have any effect on my prints). 

**MY OPINION**: Ignore it. And as for Mac users, they probably don't need to see it.

thkrmr, 2025-04-21:
@tynielsen weird Insight on your part, given that in this thread it's specifically mentioned how various calibration tests are broken on macs because of this error.

kfbest, 2025-04-26:
issue still exists in 2.3.0 on macOS

swdwi2, 2025-04-29:
Issue still there in ver 2.3.0.  Any hope of a fix ?

M4xTheMan, 2025-05-03:
Happens with version 2.3.0 under Linux too.

suineg, 2025-07-05:
Unsure if it's worth mentioning that this is still an issue, is there a plan for fixing this?

18194611, 2025-11-11:
11 Nov. 2025, still exists.

SweetieRick, 2025-11-29:
+1 Temp tower and max volumetric speed tests still have non-manifold edges, using Manjaro Linux latest build and still cannot automatically fix without a windows copy of OrcaSlicer

closed 2026-01-16 by ianalexis