OrcaSlicer/OrcaSlicer #3350: specify the part cooling fan ramp up/down lag

jlo1975, 2023-12-29:
**Is your feature request related to a problem? Please describe.**
On some material like PETG where the part fan is very low. We need high fan speed for overhang or bridge.
But we miss the correct timing as the fan part is too late when the bridge/overhang is starting and still at high rpm when the bridge/overhang is finished

**Which printers will be beneficial to this feature**
Bambu lab printers

**Describe the solution you'd like**
add a setting (in filament setup?) in part fan option to setup the lag in ms.
For ex a setting of -500 would move into the gcode 500 ms sooner the ramp up and ramp down of the part fan.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

amicloud, 2024-01-01:
This feature is in Orcaslicer, at least with the 1.9.0 beta I am currently using. 
[image]

jlo1975, 2024-01-01:
thanks !
indeed I didn't check the printer setting

jlo1975, 2024-01-01:
ok i've done some testing.
and it seem this feature does not work.
"the overhang does not wait" the correct fan speed

here a test with a 1 sec lag vs a 5 sec lag
you will see in the Gcode that the kickstart is at exactly the same location (as specified by the feature).
but there is not barrier/pause to let it ramp up and so the overhang is printed without the correct fan speed

see screenshot.

[image]
[image]
[image]

amicloud, 2024-01-02:
Not sure what could be going on there. It seems to be working for me, as shown in the bridging tests below. The first is with 1.25 second and the second with 5 second ramp up time, last is with 0.5 second

[image]

[image]

[image]

Maybe the kick-start time has something to do with it but tbh at this moment I don't have time to look into it myself

jlo1975, 2024-01-02:
in your screenshot what we see is the the theorical fan speed.
but in real life the fan is not a the shown speed when starting the overhang section
we don't see in the code something after a M106 s*** code like an extruder speed down/flow down or a barrier/waiting commang like  a G4 ***  to let the time to the fan to speed up.

you can make a real life test with a simple overhang test and put  a 2 - 3 sec in the fan setting a you will see that the printer start the overhand section without waiting the 2 sec.

I will put some screenshot of what i'm saying in few minutes (I will edit this reply)

jlo1975, 2024-01-02:
ok here the scrreenshot and we see something is really wrong

here a test with 0 speed up time
[image]

and with  2 sec speed up time and we see no diff !!!!
[image]

but something more weird if following
here you will see a kdiff3 of the generated GCODE (0sec  speed up vs 2 sec speed up)
and we can see something is missing on the 2 sec GCODE !!!! the M106 command !
[image]

jlo1975, 2024-01-02:
here the weird thing

test with 0 speed up time look at the inner  line with speed at 90%
[image]

and now with a speed time of 2 sec the inner line is no more at 90% but at 30% ...... weird no?
[image]

confirmed by the kdiff3 between the 2 gcode a M106 command is missing when we specifiy the 2 sec speed up time.
Bit still no barrier/wainting time or slow down of the hotend speed or flow to let the fan ramp up.
[image]

jlo1975, 2024-01-02:
here another test where i'm showing visually where the barrier/waiting block should be by playing with the fan kick start time (aka 100% fan). and we can see no change between the  2 gcode except where the end of kick start is

[image]

[image]

jlo1975, 2024-01-02:
finally on my last screenshot above I've just noticed -> there is a blank line just after the M106 S255 command like something (a command ?) should be here but not generated

jlo1975, 2024-02-15:
ok found out the problem is comming from the check box -> only overhangs...
more info here -> https://github.com/SoftFever/OrcaSlicer/issues/3498

jlo1975, 2024-02-15:
closing this issue but let the BUG report opened as  found out the problem is coming from the check box -> only overhangs...
for cooling fan ramp up/lag to be correctly handled for overhang you have to untick the check box -> only overhangs...
see my last comments on the bug report  https://github.com/SoftFever/OrcaSlicer/issues/3498

closed 2024-02-15 by jlo1975