OrcaSlicer/OrcaSlicer #3364: Add job cost {variable_name} to exported gCode filename format options

tlhintoq, 2023-12-30:
It would be really nice if exported gCode names could include the job cost as well as the job print time.
After much experimentation with possible variable names the only working one I could find was `filament_cost[0]` but that returns the total spool cost as set on the material  ($20/kg) instead of the cost of running this job.
[image]

tlhintoq, 2024-01-17:
Keep alive - Still an issue in 2.0 developer/nightly

tlhintoq, 2024-02-08:
Hey bot, please don't close me - this is still an issue.

tlhintoq, 2024-03-13:
Problem still exists in current 2.0 nightly as of 13mar2024

tlhintoq, 2024-04-15:
Keep alive.  Problem still exists in current 2.0 nightly as of 15apr2024

tlhintoq, 2024-04-23:
@Noisyfox @SoftFever 
Just wondering... Any sense of whether this will get on the roadmap?  or is this maybe subsumed into another ticket?  I seem to recall someone was working an entire framework of costing.  Any chance that those variables are available for file naming?
[image]

SoftFever, 2024-04-23:
@tlhintoq 
Hmmm it should be supported already.
Can you try this variable name instead? 'total_cost'
Let me know if it works

tlhintoq, 2024-04-23:
>  Can you try this variable name instead? 'total_cost' Let me know if it works

Yes.  Thank you.  `{total_cost}` does the job.  Formatted a little long, but I can live.  Just wondering... I where can I find a proper list of usuable variables?  I've asked on Discord before getting banned, I've asked on facebook.  The only answer was "same as bambu slicer" which still isn't a location.  If I could have found a complete list I wouldn't have had to raise this as a bug and  then pester people for 9 months which nobody wants.
[image]

closed 2024-04-23 by tlhintoq