OrcaSlicer/OrcaSlicer #3366: Ender 3v2 layer height presets

Egosumumbravir, 2023-12-30:
### OrcaSlicer Version

1.8.1

### OS version

Windows 10

### Printer

Creality Ender 3v2

### How to reproduce

Setup the stock 0.4mm nozzle Ender 3v2.
Try for the optimal 0.16mm layer height profile - it's 0.15mm.

### Actual results

Cheap Creality printers give their best results when working within magic numbers - one of these is layer heights where 0.04mm is the best divisor as it's a whole step on the axis.
0.24, 0.20, 0.12 & even 0.08 if you're patient.
0.15 needs to be bumped to 0.16 to fit in the magic layer height number scheme.

Two other printers seem to have the same error in the system/Creality/Process folder CR10Max & Ender5Pro (2019).
They should all be 0.16 as per all the other 0.16mm printer profiles.

### Expected results

Layer that sit on whole divisors of Creality stepper motors.

### Project file & Debug log uploads

[Orca_log.zip](https://github.com/SoftFever/OrcaSlicer/files/13797866/Orca_log.zip)
[Ender_0.15mm_Layer.zip](https://github.com/SoftFever/OrcaSlicer/files/13797867/Ender_0.15mm_Layer.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bistory, 2024-01-02:
@SoftFever this is a simple fix, but what is your policy regarding that issue ? Simple create a new setting for 0.16 and don't touch 0.15 even if it's faulty/legacy ?

SoftFever, 2024-01-03:
What was the issue exactly? The process is named as 0.16 mm but the actual layer height is set to 0.15?
If so, I agree it's a bug. Changing the value(not the name) is sufficient.

If it's correctly named as 0.15 process then it's not a bug.
I'm open to add a new 0.16 process profile in this case if anyone is willing to help.

Egosumumbravir, 2024-01-03:
> What was the issue exactly?

0.15 name and height. Which on Enders is a microstep on the Z motor and gives less than ideal print quality, especially if using z-hopping.

0.16 is a full step, a "Magic Number" for Enders steppers and lead screws and much more accurate than a microstep.

It would be best to revise these three incorrect profiles, not everyone will be aware that 0.15 is a bad choice (or why).
Alternatively, make new 0.16 profiles and remove the 0.15s?

Happy to help, but need to learn how first!

bistory, 2024-01-03:
> What was the issue exactly? The process is named as 0.16 mm but the actual layer height is set to 0.15? If so, I agree it's a bug. Changing the value(not the name) is sufficient.
>
> If it's correctly [...]

There is a 0.15 profile using 0.15 layer height and as @Egosumumbravir told, it's not a magic number, so this can lead to layers overlappings.
The issue comes from Creality which has 0.15 profiles in their slicer...
The clean fix would be to remove 0.15 profiles and replace them by 0.16 but I know you are against removing something in the profiles in case someone uses it.

SoftFever, 2024-01-05:
I see, we can add a new 0.16 profile but should keep the 0.15 profile

bistory, 2024-01-11:
I made some researches @Egosumumbravir 
Prusa made a calculator where you can find the "magic numbers" and it seems that 0.15mm is the correct number for that printer : https://blog.prusa3d.com/calculator_3416/

Egosumumbravir, 2024-01-12:
As did I @bistory 
https://www.youtube.com/watch?v=WIkT8asT90A
https://makershop.co/ender-3-magic-numbers/
https://3dprinterly.com/3d-printer-magic-numbers-getting-the-best-quality-prints/
https://howto3dprint.net/ender-3-magic-numbers/

The Prusa calculator seems to think the step length is 0.01mm, instead of what everyone else is saying which is 0.04mm.

SoftFever, 2024-01-12:
It depends on what leadscrew/belt pitch you use in the Z axis system + "step per Revolution" of the step motor + gear ratio.

In modern system, we all use micro steps, I would suggest don't bother with so called magic numbers ;)
There is convincing evidence to support this theory.

bistory, 2024-01-12:
> It depends on what leadscrew/belt pitch you use in the Z axis system + "step per Revolution" of the step motor + gear ratio.
> 
> And don't forget in modern system, we all use micro steps, I would suggest don't bother with so called magic numbers ;) There is no convincing evidence to support this theory.

I agree with you 👍

Egosumumbravir, 2024-01-12:
> It depends on what leadscrew/belt pitch you use in the Z axis system + "step per Revolution" of the step motor + gear ratio.

Absolutely. Creality standard hardware is very consistent across their product line.

> And don't forget in modern system, we all use micro steps

Indeed true. The whole point of the "magic number" is full steps are inherently more accurate than a microstep. Most of the time on other axes this is imperceptible, but the regular spacing of Z reveals inaccuracy.

Actually, this is a good point - if this is true, I should be able to show it empirically with a z-wobble tower sliced @ 0.16 and 0.15. Hold on, I should be able to run this over the weekend and report back the results.

Egosumumbravir, 2024-01-19:
[image]

Alrighty, empirical results.
200mm vase towers printed back to back. Same settings, same filament dried 6h @ 48°C and fed from a sub-5% drybox. Only parameter changed is layer height - 0.16 v 0.15. Dual-colour silk filament to really show off any problems.
Creality v4.2.7 board, MRiscoC current version firmware loaded with all the goodies - input shaping, pressure advance, MPC.
For future toyfund noting: Capturing fine details without a DSLR camera is darn tricky. Sensor noise on a phone, even a nice one makes it difficult to discern whats dogs bollocks IRL.
Close examination shows I have a little z banding to sort out on this machine, even at the "magic numbers"; however, slicing at a muggle number amplifies regular banding and bulging layer frequency issues significantly.

0.15 will pass the three foot test. But at close examination, the 0.16 wins for precise layer stacking.

closed 2024-04-27 by github-actions[bot]