OrcaSlicer/OrcaSlicer #3367: Easy plate naming

Nevio024, 2023-12-30:
**Is your feature request related to a problem? Please describe.**
Naming all the different plates of a big project is a pain. It's just unnecessarily complicated.

**Which printers will be beneficial to this feature**
This would make using and naming multiple build plates for projects in Orca Slicer easier.

**Describe the solution you'd like**
Just a simple click on the "Untitled" (or on a symbol next to it) would allow the title to being edited. It could be confirmed with Enter or just by clicking somewhere else. There would be no extra window opening. The title should be edited in the standart "Prepare" view.

Eldenroot, 2023-12-30:
Bambulab studio 1.8.2 contains this feature. There is an icon of pencil next to the plate name, you can easily rename it.

Nevio024, 2023-12-30:
Yes, exactly! Having that in OrcaSlicer would be awesome!

Eldenroot, 2023-12-30:
Yes, I agree!

This is the screenshot from Bambulab studio 1.8.2

[image]

@SoftFever - please, could you port this?

Noisyfox, 2024-01-03:
> Bambulab studio 1.8.2 contains this feature. There is an icon of pencil next to the plate name, you can easily rename it.

But that still shows a popup instead of editing the name in place.

SoftFever, 2024-01-03:
They added an icon when they ported this feature from Orca. I think I can add a behaviour that invokes the plate name dialog when double clicking the plate name

Eldenroot, 2024-01-14:
@SoftFever - this would be awesome, together with sync up to Bambu studio v 1.8.3

PR #3756 (2024-01-21): Add plate name edit btn
Fix #3367

[image]
closed 2024-01-21 by SoftFever