OrcaSlicer/OrcaSlicer #3373: Crilality K1 - Air filtration option didn`t work

Gboy1995, 2023-12-30:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Printer

Criliaty k1

### How to reproduce

1. Go to Filament Settings
2. Cooling
3. Exchaust fan
4. Check - Active air filtration
5. 5% fan speed on start and 100% on end
7. Export the G Code
8. Inspect G code file - M106 command

### Actual results

The G code command to activate the chamber fan is named with P3 fan, but K1 didn`t have P3 fan, only P0, P1 and P2.
G Code line: M106 P3 S255
[image]
[image]

### Expected results

The Fan variable seted must be P1 instead of P3
For the start settings: M106 P1 S25
For the end settings: M106 P1 S255

### Project file & Debug log uploads

[apoio_2_ABS_10m12s.zip](https://github.com/SoftFever/OrcaSlicer/files/13798939/apoio_2_ABS_10m12s.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sheinbergon, 2023-12-30:
@SoftFever I second this issue. exhaust fan is actuall P1, not P3 for the K1 printer. Looking at the code, is there a way to override the mapping per profile?

Guilouz, 2023-12-31:
It's just Creality that didn't configure its configuration correctly.
I fixed this on my Wiki : https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Improve-Fans-Control

Just add this in your M106 macro:

{% if params.P is defined and params.P|int == 3 %}
  {% set fan = 1 %}
{% endif %}

Before:

SET_PIN PIN=fan{fan} VALUE={value}

closed 2024-04-08 by github-actions[bot]

kemsky, 2024-09-07:
I don't see why it can not be done in the slicer, it is much easier than changing all macros that already use m106