OrcaSlicer/OrcaSlicer #3375: Consistent crash

WimVDK, 2023-12-30:
### OrcaSlicer Version

1.9.0 Beta1

### OS version

Windows 11 23H2 build 22631,2861

### Additional system information

13900k (non-oc), RTX Strix 4090 OC, 64GB (4*16gb) Corsair dominator Titanium at 4533Mhz (XMP won't boot)

Reguraly with both OrcaSlicer & BambuStudio i get an STL i can consistently try to slice and make the slicer crash. If i try too much it sometimes takes the whole system with it (with BSOD's like Clock Watchdog Timeout). No matter what settings i try to change, doesn't help. Currently experiencing with with the "AMS Pro 1.00-full-set-petg.3mf" file from Humebeam @ printables.com. The same applies to BambuSlicer so it's something that's shared between them, but OC is my preferred slicer due to more control settings so i figured i'd report here.

There is a workaround which is to rightclick the desktop shortcut and set it to run in Windows 8 compatibility mode. It's slower to slice, but in that mode i've yet to experience a crash. Attached are the logs for just a few of the slice tries

### Printer

Bambulab X1C

### How to reproduce

1. Load 3mf file
2. slice
3. Crash

### Actual results

Full system crash bsod depending on the circumstances

### Expected results

succesful slice in non-windows 8 compatibility mode

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13799008/log.zip)
[hydra-pro-1.00-full-set-petg.zip](https://github.com/SoftFever/OrcaSlicer/files/13799009/hydra-pro-1.00-full-set-petg.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

silence48, 2023-12-31:
There is a likely relationship to issue [2921](https://github.com/SoftFever/OrcaSlicer/issues/2921)

WimVDK, 2023-12-31:
Very possible, I do remember seeing that ACCESS VIOLATION bsod a few times too, and the hydra ams parts do have a lot of layers (at .2mm layer height). If there's anything I can further provide to help fix, I just need to know what.

silence48, 2023-12-31:
Likely related to issue [2921](https://github.com/SoftFever/OrcaSlicer/issues/2921) also.  I've left some comments there regarding it.

silence48, 2023-12-31:
> Very possible, I do remember seeing that ACCESS VIOLATION bsod a few times too, and the hydra ams parts do have a lot of layers (at .2mm layer height). If there's anything I can further provide to help fix, I just need to know what.

Thank you for your insight.  I just actually found this project, in fact because of this bug (the bug also exists in bambu slicer) And so I would rather contribute to this than a for profit company, so here I am 😆 

I will definitely report back if I find a solution or can enact one in the code.

silence48, 2023-12-31:
> Very possible, I do remember seeing that ACCESS VIOLATION bsod a few times too, and the hydra ams parts do have a lot of layers (at .2mm layer height). If there's anything I can further provide to help fix, I just need to know what.

can you run a test and enable compatibility mode for windows 8 and then try again? For me this caused it to no longer crash, and instead it is froze up but it says it's using 3% of my cpu and over 8gb of memory currently.  I'm gonna let it sit for a while to see if anything interesting pops out my logs.

silence48, 2023-12-31:
It just now finished slicing the file.  

Try to enable windows compatibility mode (for windows 8) And maybe it'll help you like it helped me!
[image]

SoftFever, 2023-12-31:
Hey everyone,

Just a heads-up - OrcaSlicer tends to crash if we switch to the device page too soon, before our printer's connected. Good news, though – I've got a fix lined up for the next update. For now, just hang on for about 10-30 seconds after firing up OrcaSlicer before you switch to the device page.

Also, if anyone's game, I'd really appreciate some help testing this fix. The latest build will be ready in about an hour. You can download it [[here](https://github.com/SoftFever/OrcaSlicer/actions/runs/7368627968)] and let me know if you run into any more crashes. Thanks a ton!

WimVDK, 2023-12-31:
@silence48 as mentioned at the bottom of the OP 8 already am. Even CM mode isn't 'safe' from the crashes.

@SoftFever 

`For now, just hang on for about 10-30 seconds after firing up OrcaSlicer before you switch to the device page.`

I don't see how that's going to assist, as my OC doesn't switch to device tab at all and my crashes were during slicing, generating gcode, uploading to printer. I never hit 'send and print'. I only use 'send', and launch a print from the X1C screen.

In bed now, will try the new build tomorrow

WimVDK, 2023-12-31:
@SoftFever As expected, this recent build did NOT fix the crashes during slicing. Still crashes during slicing (always on plate 2 of this 3md file) before you even get to send the sliced to the printer

WimVDK, 2023-12-31:
And i just had it crash (in compatibility mode) trying to slice 4 simple PTFE buttons for the AMS. Crashes when trying to use 5 bottom & top layers.
[PTFE load buttons.zip](https://github.com/SoftFever/OrcaSlicer/files/13801835/PTFE.load.buttons.zip)

SoftFever, 2024-01-01:
Looks like it's a different crash issues.
Let me take a look.

SoftFever, 2024-01-01:
It's a regression bug cause by PR https://github.com/SoftFever/OrcaSlicer/pull/3014

WimVDK, 2024-01-01:
What i have noticed running the latest build is that the crashes with those PTFE buttons sometimes do slices, but it fails to send. When hitting send it goes right for about the first 10% and when it should actually start uploading, OS crashes even in compat mode. Reverting to 1.8.1 fixed the upload issue but it doesn't fix the slicing crashes.

SoftFever, 2024-01-01:
Can you try this build? 
https://github.com/SoftFever/OrcaSlicer/actions/runs/7377220740

WimVDK, 2024-01-01:
Certainly, if you can tell me how to download the windows EXE from it :)

SoftFever, 2024-01-01:
> Certainly, if you can tell me how to download the windows EXE from it :)

Sure, you can download it here: https://github.com/SoftFever/OrcaSlicer/actions/runs/7377220740
[image]

WimVDK, 2024-01-01:
Still crashes the moment i try to slice plate2 of the initial 3mf i posted (hydra pro AMS)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13804934/log.zip)

WimVDK, 2024-01-06:
@SoftFever I see there's been an official 1.9.0 release. I don't suppose it this issue does it?

SoftFever, 2024-01-08:
> @SoftFever I see there's been an official 1.9.0 release. I don't suppose it this issue does it?
> 
> _Edit_ Nope. Still crashes, even on something as simple as https://www.thingiverse.com/thing:5240529
> 
> Project file & logs attached.
> 
> [files.zip](https://github.com/SoftFever/OrcaSlicer/files/13850482/files.zip)

I have tried all the projects you attached, but I can't reproduce the crashing issue. Your case seems different from any other reports. There might be some conflicting environment settings in your OS. I have no clues at the moment.

WimVDK, 2024-01-08:
If you wanna do a TeamViewer session and have a look around, that certainly can be arranged. BS seems to work for the moment.

chuckie1992, 2024-01-11:
Have you guys tried changing the affinity of the process. I have crashes in every slicer on my main rig which contains a i9 13900k. Soon as i set the affinity to only core 0-7(The first cores are P cores so simple to reproduce) all crashes have been stopped in every slicer i tried so far.  I have a i7 11700k and on that machine no crashes since it does not have the new hybrid tech from intel.

WimVDK, 2024-01-11:
Care to tell me how I can do that?

chuckie1992, 2024-01-11:
@WimVDK Like this
[image]
[image]

WimVDK, 2024-01-11:
No more crashes then? Even without setting compatibility mode to win 8?

chuckie1992, 2024-01-11:
Yes

WimVDK, 2024-01-11:
Then it's certainly worth a try after work this evening. Thanks for the tip. Will report back

WimVDK, 2024-01-11:
@SoftFever User @chuckie1992 may be on to something. I just succesfully sliced the original attached project file. A bambu Tidy project with 28 different buildplates, and Fibo spool holder project with 6 plates (1 more than the project in this discussion). They may be on to something. Even with my 13900k, cpu usage spiked upwzrds of 40% and depending on the plate into the lower 50's but they didn't crash.

@chuckie1992 i didn't have 0-7 as you mentioned, but i did have 0-4 as the screenshot. They seem to do the trick, i'll keep monitoring this over the next few weeks.

chuckie1992, 2024-01-11:
> @SoftFever User @chuckie1992 may be on to something. I just succesfully sliced the original attached project file. A bambu Tidy project with 28 different buildplates, and Fibo spool holder project [...]

Yea this is 100% the reason Slicers like OrcaSlicer crash. They are not programmed to support E cores.
You also should have 8 cores with your 13900k, just scroll down in the Affinity list. 
You also should be able to select core 0 to core 15 due to Hyper Threading technology on intel cpu's. The first 15 cores are P cores and its threads. I just tested with the 1st 8 cores and it stopped crashing.

WimVDK, 2024-01-11:
@chuckie1992 I got 32 cores (0-31). It is a lot faster with 0-15.

Now just need to find a way to set process affinity permanently, cause it seems to lose that configuration once the application is closed.

@SoftFever Could you make this a feature please? To restrict OrcaSlicer to P cores only?

chuckie1992, 2024-01-11:
> @chuckie1992 I got 32 cores (0-31). It is a lot faster with 0-15.
> 
> Now just need to find a way to set process affinity permanently, cause it seems to lose that configuration once the application is closed. Doing it the shortcut way as described here https://answers.microsoft.com/en-us/windows/forum/all/how-to-permanently-set-priority-affinity-with/4e83fd39-34a7-49fe-a54a-ee891c38b737 doesn't work. IT launches a command window where you still manually need to enter "orca-slicer.exe"
> 
> @SoftFever Could you make this a feature please? To restrict OrcaSlicer to P cores only?

Yes an app called process lasso: https://bitsum.com/
With that you do the same as in Task manager but the Affinity stays after reboot they claim. And they have a feature to disable Efficiency mode so nothing will run on E cores. But i would just set affinity manually.

And i got 32 cores too 8P cores+ 8 Threads=16+16 E cores= 32 cores
0-15 are P cores/P core threads 16-31 are E cores(no hyperthreading on those so only single thread per core)

@SoftFever Just have to find what exactly in his code doesn't run well under E cores and try to solve it. It should be able to use those cores too to improve overall performance in the app.

chuckie1992, 2024-01-11:
You also could try this article, Similar to what Microsoft explains but uses Hexadecimals
https://www.windowsdigitals.com/how-to-set-process-cpu-affinity-priority-permanently-in-windows-10/
Windows 10/11 are basically the same. From what i've heard basically GUI refresh and changes to Scheduler to support the new intel hybrid cpu's. But easiest route would be Process Lasso. Free version should do everything you need it to do.

WimVDK, 2024-01-11:
I'll look into it, I'll also report this behavior on the Bambu slicer GitHub tomorrow. Both are based in a way on prusa slicer, but afaik that one doesn't crash?

chuckie1992, 2024-01-11:
Maybe they did update the code to support E cores. I own a K1 and had to look for other slicers since i keep having to use my wife her PC to slice things everytime since she has my old rig with an i7 11700k in it. Not the biggest issue but just an annoying workflow to keep swapping over to do something :D.
Creality Print has the same issue and solved with the same method(did report this to them via e-mail and told them what is causing it). I haven't tried Bambu slicer yet myself. But basically it is happening across alot of slicers out there with these new hybrid cpu's

WimVDK, 2024-01-11:
2nd link didn't work either. A quick test of lasso does for the moment. Richtclick > set affinati > Always > select cores > P cores only. Close Lasso, don't close background engine. Affinity persists through program restarts. Have yet to try a system reboot. I don't like doing those during the week. My system is under normal circumstances running 24/7.

Reporting this to bambuslicer github this weekend and ask on prusa if they did anything particular in regards to E-cores. Maybe they can show the relevant pull requests.

chuckie1992, 2024-01-11:
> 2nd link didn't work either. A quick test of lasso does for the moment. Richtclick > set affinati > Always > select cores > P cores only. Close Lasso, don't close background engine. Affinity [...]

And i tested after reboot, They are indeed persistent so 1 time setting it up with Process Lasso and no more headache :)

WimVDK, 2024-01-11:
Just found a project for multicolor (need 2 AMS), somehow I still manage to make it crash lol. I'll tinker with that over the weekend. Seems to crash around infill toolpath for the moment

SoftFever, 2024-01-12:
@WimVDK @chuckie1992 
It's unlikely the P/E core issue as I'm using Intel 13900K too.
It might be a RAM issue as you are slicing an complex model.
How much RAM do you have?

WimVDK, 2024-01-12:
64gb it ain't a ram issue. Even the FIBO crashed at barely 3.5gb used, ditto for 28 plate project. They sliced fine when disabling the E cores through affinity when they would otherwise crash.

The current one (multicolor dragon) I'm trying to slice, slices fine in 1 color but currently crashes in multi, trying to figure that one out atm

crckdns, 2024-01-14:
can confirm, slicing 20MB complex file (simplified from 200MB) would crash on i7 13700k (8C+8c) and 32GB ddr5 with following system-freeze.

enabling ONLY the cores 0-15 (basically only P cores) runs trough without problems.

WimVDK, 2024-02-18:
still happens

closed 2024-06-05 by github-actions[bot]