OrcaSlicer/OrcaSlicer #3382: Unable to load user profiles

Knuckleheadx, 2023-12-31:
### OrcaSlicer Version

1.8.1

### OS version

windows 10

### Printer

Creality K1 Max

### How to reproduce

Would open the edit tab and select import config. Configs would be selected and loaded with no issues.
Once I add the wire less connection to the K1 Max all User profiles are gone.
Attempt to reload configs and it states one is already there would you like to replace it.
If click yes to replace the prompt states one profile has been uploaded.

### Actual results

Unfortunately no profiles has been uploaded. 
Completely uninstalled orca slicer and profiles to try again. Before adding the wire less connection all user profiles are added with ease and show up. Once the wireless connection is added they disappear and cannot be found. While trying to reload saved configs the prompt claims they already exist. If to override prompt it will say 1 had been uploaded with no change.

### Expected results

Should be able to load the config and it would show up just as it did before I added the ap address for wifi

### Project file & Debug log uploads

[0.20mm Quality @K1 Max.json](https://github.com/SoftFever/OrcaSlicer/files/13799513/0.20mm.Quality.%40K1.Max.json)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bistory, 2023-12-31:
Can you try the latest build here and see if the bug still occurs ? => https://github.com/SoftFever/OrcaSlicer/actions/runs/7369939825
It looks like the bugs I fixed with that pull request

Knuckleheadx, 2023-12-31:
Gave it a shot and unfortunately still no user presets. Also when I try to load them it still says i already have them. Even with an override all it appears I'm still having the same issue.

closed 2024-04-08 by github-actions[bot]