OrcaSlicer/OrcaSlicer #3392: Hole wall problem

olcayoren, 2023-12-31:
### OrcaSlicer Version

1.8.1

### OS version

Macos Sonoma 14.0

### Additional system information

Macbook air m1 2020 arm

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

- Drawing a hole in the wall

https://github.com/SoftFever/OrcaSlicer/assets/33813227/3e794160-d594-4f3f-be0f-21da19c782da

[image]
[image]

### Actual results

- Internal overflow problem in a drawing with a hole in the wall
- There is no problem reflected on the outer surface

### Expected results

None

### Project file & Debug log uploads

[CaseV2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13802029/CaseV2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

wrenchonline, 2024-01-10:
the same problem

wrenchonline, 2024-01-12:
Arachne mod is enabled by default and seems to have nothing left to do with it.It doesn't seem like a universal solution

wrenchonline, 2024-01-12:
> > the same problem.I have looked for many situations, but only this [website](https://forum.prusa3d.com/forum/prusaslicer/mitigate-wall-bulge/paged/1/) has some help, it seems to be an error in the Prusa kernel.
> 
> From what I've read, they recommend using the arachne mod.

Arachne mod is enabled by default and seems to have nothing left to do with it.It doesn't seem like a universal solution

olcayoren, 2024-01-18:
> I changed the line width and speed of the inner and outer walls, and there has indeed been an improvement
> 

Which values did you write? Can you post a picture?

wrenchonline, 2024-01-19:
yes，But I found that it's not the main reason, I probably know the main reason now.

wrenchonline, 2024-01-19:
> > I changed the line width and speed of the inner and outer walls, and there has indeed been an improvement
> 
> Which values did you write? Can you post a picture?

yes，But I found that it's not the main reason, I probably know the main reason now.

[image]

wrenchonline, 2024-01-19:
> > I changed the line width and speed of the inner and outer walls, and there has indeed been an improvement
> 
> Which values did you write? Can you post a picture?

This can greatly reduce the layering, which seems to be faintly visible.

wrenchonline, 2024-01-19:
> > > I changed the line width and speed of the inner and outer walls, and there has indeed been an improvement
> > 
> > 
> > Which values did you write? Can you post a picture?
> 
> This can greatly reduce the layering, which seems to be faintly visible.It seems that there are more available variables to handle, and the feel will also be much worse after reducing precision. good luck for your.

olcayoren, 2024-01-19:
> I changed the line width and speed of the inner and outer walls, and there has indeed been an improvement
>
>
> Which values did you write? Can you post a picture?
>
>
> This can greatly reduce the [...]

Thanks, I hope I can find a permanent solution.

wrenchonline, 2024-01-20:
> Thanks, I hope I can find a permanent solution.

I have discovered a new solution that can reduce the layering without losing the outer wall.
[image]

wrenchonline, 2024-02-19:
I have found that there is another issue that leads to hole wall problems, which is the shrinkage rate of consumables, which is the difference in heat dissipation and time between different layers during bridging.Use OrcaSlicer Layter time to view the printing time of different layers. Some layers print too slowly and dissipate heat too quickly, causing consumables to shrink due to stress.

olcayoren, 2024-02-20:
Thanks for your help, yes, in most filaments, reducing the wall speed and temperature causes healing to the holes in the wall. Then, in such perforated models, it is necessary to follow the layer time chart as much as possible. It seems that playing with the minimum layer time value may also be beneficial in terms of cooling.

closed 2024-05-29 by github-actions[bot]