OrcaSlicer/OrcaSlicer #3395: Model Not Visible When Painting Seams or Supports

XSvFury, 2023-12-31:
### OrcaSlicer Version

1.9.0 Beta

### OS version

Windows 10

### Additional system information

AMD Ryzen 4800HS, 16 GB memory, NVidia GTX 1660 Ti Max-Q

### Printer

Voron Trident, Ratrig V-Core 3, Bambu P1S, Creality K1

### How to reproduce

1. Place an stl model onto the bed.
2. Select either support painting or seam painting.

### Actual results

Model disappears. The mouse will still change to the painting tool when it is over where the model is supposed to be.

Note: I have used more than few versions of OrcaSlicer on this computer and this has never been an issue. Also, I reinstalled version 1.8.1 Release and it works fine.

### Expected results

The model should stay so it can be painted.

### Project file & Debug log uploads

[desk_tool_holder-prject-file.zip](https://github.com/SoftFever/OrcaSlicer/files/13802213/desk_tool_holder-prject-file.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13802215/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2024-01-03:
Have you try running the app using iGPU/eGPU?

XSvFury, 2024-01-04:
> Have you try forcing running the app with iGPU/discrete GPU?

I tried both and both had the same result.

michaltulach, 2024-01-06:
I have the same problem and filament painting doesn´t work too.

XSvFury, 2024-01-06:
> I have the same problem and filament painting doesn´t work too.

Thank you for posting. Hopefully it gets fixed soon.

JohnGames, 2024-01-09:
Reference this issue for temporary work around: https://github.com/SoftFever/OrcaSlicer/issues/3487

PR #3663 (2024-01-15): Fix painting Gizmo rendering on AMD Vega GPUs
Fixes #3395
Fixes #3487

Also fixes an issue that seam & support painting gizmo render improperly when wireframe is enabled:
The wireframe is displayed properly in MMU gizmo:
[image]
But looks like this when switching to seam/support gizmo:
[image]
closed 2024-01-15 by SoftFever