OrcaSlicer/OrcaSlicer #3408: Integrate Scarf Seams by MichaelPlatypus

Manpoozle, 2024-01-01:
This feature was added as a Pull Request to PrusaSlicer [Here](https://github.com/prusa3d/PrusaSlicer/issues/11621).

This Feature Request is to pull the code directly into OrcaSlicer, skipping PrusaSlicer, given that it generally takes quite a while for a new feature to get picked up in PrusaSlicer and there is a lot of community excitement around it.

Here is a quick summary on this reddit thread.
https://www.reddit.com/r/prusa3d/comments/18a6dv8/a_general_method_for_better_seams/

Noisyfox, 2024-01-01:
I couldn't find the PR for this.

Manpoozle, 2024-01-01:
I apologize, the PR is specific to scarf seams in vase mode, and though MichaelPlatypus has uploaded a gcode file of a non-vase mode version, I don't see an implementation either.

https://github.com/b3n3d1k7/PrusaSlicer/commit/173a69391e3091cdb63dd9073a2e5bdf5818ea95

MichaelJLew, 2024-01-04:
What is a "PR"? 

Given the adverse comments that I have received on Reddit and GitHib it feels a bit like only the people who have full mastery of C coding and all of GitHub are able to contribute to open-source software. If so, then that is very unfortunate.

Manpoozle, 2024-01-04:
@MichaelJLew I'm not sure exactly what you mean. A PR is a "Pull Request" and it is a code patch that is published for review from other members of a codebase. It's commonly used among software engineers because of the universality of Git (from which GitHub derives its name), and every change a software engineer makes day in and day out is published through pull requests. So you can imagine if you're talking about 5 of these a day among a team, it quickly falls into an abbreviation that becomes common parlance across the industry. If you want to learn more, of find other points of confusion, you can ask these kinds of questions of something like ChatGPT, and you'll be amazed how effectively it (and other large language models) can help in the software realm.

As to your comment about C, I'm still not 100% sure what you're saying. I haven't looked into the codebase that backs Orca. The creators would have likely chosen to stay with the language(s) chosen by the developers beneath them in the stack (Slic3r, Prusa Slicer, Bambu Studio, Super Slicer). If that's C, then it's a product of the times, where the language choice predates some of the other options a developer might choose today, like Rust. I haven't looked though, and I suspect it's not C because UIs are a nightmare in C. But the core slicer might be for speed.

Projects are in the languages they are in because of a complex multi-factor decision making process and history. Nobody _wants_ to write code in C in 2024. But rewriting projects this large and complex, especially without being paid, and especially then releasing them for free is very admirable. There is a moment coming only a couple, maybe a handful, of years out where the AIs will happily convert this entire project to any language you like. They already write the code better than an intern, and they'll explain to you. You can contribute, and if you can't quite yet, you will be able to soon.

Noisyfox, 2024-01-26:
I think this is the same idea as #3211

closed 2024-03-02 by SoftFever