OrcaSlicer/OrcaSlicer #3413: Ensure vertical shell thickness off issue with Bridge anchor area

qwewer0, 2024-01-01:
### OrcaSlicer Version

1.9 Beta/Alpha, 1.8.1, 1.8

### OS version

Windows 10

### Printer

Ender 3

### How to reproduce

1. Open Slicer (1.9 Alpha/Beta, 1.8.1, 1.8 or pull request [3285](https://github.com/SoftFever/OrcaSlicer/pull/3285) by igiannakas)
13. Open the project file
14. Turn off the `Ensure vertical shell thickness` setting
15. Slice plate
16. Go to layer 82
17. See the bridge anchor area

The last version was 1.7 when it worked normally.

### Actual results

The ends of the bridges are only anchored to the part with a single wall thickness.
[image]

### Expected results

The ends of the bridges should be anchored to the part with the number of wall lops. (or at least with more than 1 wall thickness)
[image]

### Project file & Debug log uploads

[Cube Bridge.zip](https://github.com/SoftFever/OrcaSlicer/files/13804526/Cube.Bridge.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-01-01:
In my PR above have you tried with the new option enabled? Further reduce solid infill?

qwewer0, 2024-01-01:
> In my PR above have you tried with the new option enabled? Further reduce solid infill?

Yes, but it's the same with or without.

igiannakas, 2024-01-01:
Reproduced; however this is not a bug - it's the legacy behaviour which is triggered when ensure vertical shell thickness is turned off. Not sure if this needs a fix or not- I printed a similar bridging test and the bridges were successful.

qwewer0, 2024-01-02:
> Reproduced; however this is not a bug - it's the legacy behaviour which is triggered when ensure vertical shell thickness is turned off. Not sure if this needs a fix or not- I printed a similar bridging test and the bridges were successful.

My printer is in pieces right now, but I too will test it and close this if it works just fine.

qwewer0, 2024-01-02:
I did a few slicer tests - printer is still broken - and found something interesting:

I cut down the cube in the project file to only have a single layer at the bridging area. [Cube Bridge - Cut.zip](https://github.com/SoftFever/OrcaSlicer/files/13810855/Cube.Bridge.-.Cut.zip)
Then checked with no top and 1 or 2 bottom layers, with `Ensure vertical shell thickness` still turned on.

As we can see below, a single bottom layer will cause the same problem with the bringing anchor area.
So, while having the `Ensure vertical shell thickness` turned off does change the default bridging behavior, it isn't the main cause of the problem.

Single bottom layer:
[image]

Two bottom layers:
[image]

qwewer0, 2024-01-07:
Did a test print and it not only looks fine but it's strong too, so there is nothing to worry about.
[image]

While I still think that this slicing inconsistency will need to be checked in the future, for now, it works just fine.

@igiannakas Thanks for the check-in, and for being awesome. :star2:
Keep it up, and have a great day.

closed 2024-01-07 by qwewer0