OrcaSlicer/OrcaSlicer #3423: Can’t Add Negative Part

JeffreyOI, 2024-01-02:
### OrcaSlicer Version

1.9.0-beta

### OS version

macOS Sonoma 14.1.2

### Printer

Bambu X1C but it’s a gui issue

### How to reproduce

1. Add Primitive (Cube, e.g.).
2. Right-click primitive-> Add Negative Part

### Actual results

No object is created, size xyz values are nan.
[image]

### Expected results

A negative object should be created.

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13807112/Cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2024-01-02:
I can reproduce this issue. Will take a look soon.

qwewer0, 2024-01-02:
Can confirm.
But as a quick "fix", you can add a second negative part that will appear correctly - at least it did for me.

closed 2024-01-02 by SoftFever

CROKOBACON, 2024-09-04:
+1 i have this same issue

Noisyfox, 2024-09-04:
> +1 i have this same issue

Any step/file to reproduce?

CROKOBACON, 2024-09-04:
> > +1 i have this same issue
> 
> Any step/file to reproduce?

[bugfile_orca.zip](https://github.com/user-attachments/files/16860281/bugfile_orca.zip)

here you go, latest orca version (OrcaSlicer V2.1.1 Official Release), windows 10, Bambu Lab x1c

Noisyfox, 2024-09-04:
> > > +1 i have this same issue
> > 
> > 
> > Any step/file to reproduce?
> 
> [bugfile_orca.zip](https://github.com/user-attachments/files/16860281/bugfile_orca.zip)
> 
> here you go, latest orca version (OrcaSlicer V2.1.1 Official Release), windows 10, Bambu Lab x1c

Your file seems to be a different problem though. The negative part does not work properly, but the negative part is added successfully to the object.

PR #3431 (2024-01-02): Fix NaN when adding negative part (#3423)
Fixes #3423