OrcaSlicer/OrcaSlicer #3427: Small error in german translation

sboehle, 2024-01-02:
### OrcaSlicer Version

release 1.9

### OS version

Windows 11

### Printer

Voron 2.4

### How to reproduce

File https://github.com/SoftFever/OrcaSlicer/blob/release/v1.9/localization/i18n/de/OrcaSlicer_de.po

Row 494/495

msgstr[0] "%1$d Verbinder ist außerhalb des Schnittkontur"
msgstr[1] "%1$d Verbinder sind außerhalb des Schnittkontur"

"des Schnittkontur" is wrong

"der Schnittkontur" is correct

### Actual results

msgstr[0] "%1$d Verbinder ist außerhalb des Schnittkontur"
msgstr[1] "%1$d Verbinder sind außerhalb des Schnittkontur"

### Expected results

msgstr[0] "%1$d Verbinder ist außerhalb der Schnittkontur"
msgstr[1] "%1$d Verbinder sind außerhalb der Schnittkontur"

### Project file & Debug log uploads

https://github.com/SoftFever/OrcaSlicer/blob/release/v1.9/localization/i18n/de/OrcaSlicer_de.po

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dartrax, 2024-01-02:
Correct

sboehle, 2024-01-20:
@hliebscher 

In the latest pull (https://github.com/SoftFever/OrcaSlicer/pull/3746) these errors were not corrected.

hliebscher, 2024-01-21:
fix it in new release

closed 2024-04-28 by github-actions[bot]