OrcaSlicer/OrcaSlicer #3428: Shrinkage Multicolor issue

JonnyDoe89, 2024-01-02:
### OrcaSlicer Version

V1.81

### OS version

Windows 10

### Printer

BambuP1S

### How to reproduce

1 You adjust the shrinkage of the Filament type.
2 You multicolor your object
3 You slice the project

### Actual results

The Result, its no Multicolor sliced.

### Expected results

The object should be mulitcolored.

### Project file & Debug log uploads

[Calibration 0.4 Nozzle 20x20x10 0,8.zip](https://github.com/SoftFever/OrcaSlicer/files/13808795/Calibration.0.4.Nozzle.20x20x10.0.8.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13808837/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

elax46, 2024-01-02:
no files required

flodooby, 2024-02-01:
I have same issue with OrcaSlicer Version V1.9.0

closed 2024-05-09 by github-actions[bot]

kolyan4g, 2024-05-29:
I have same issue with V2