OrcaSlicer/OrcaSlicer #3430: Support SD card remote browsing for time-lapses and models

elax46, 2024-01-02:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Printer

Bambulab P1S

### How to reproduce

[Bambulab has released a firmware update for the p1 versions which includes support for SD card remote browsing for time-lapses and models](https://wiki.bambulab.com/en/p1/manual/p1p-firmware-release-history) but at the moment the MicroSD section returns the error initialization failed

### Actual results

Unable to view files on SD

### Expected results

view files and other additional information

### Project file & Debug log uploads

no files are required

### Checklist of files to include

- [ ] Log file
- [ ] Project file

elax46, 2024-01-02:
> Have you checked with Orca 1.9 Beta?

No I didn't check

electricar, 2024-01-02:
Same issue for me in Orca 1.9 Beta.

closed 2024-01-05 by elax46