OrcaSlicer/OrcaSlicer #3433: Crash when save Printer setting

xelarich, 2024-01-02:
### OrcaSlicer Version

1.9-beta

### OS version

Windows 10

### Additional system information

Amd Ryzen 5 1600X
Nvidia GTX 1080
8Go Ram

### Printer

Elegoo Neptune 4 

### How to reproduce

I importe Elegoo Neptune 4 Plus setting
I change the settings to correctly match the printable area of a Neptune 4
I click Save current Machine
I edit name of Preset "Elegoo Neptune 4 Plus (0.4 nozzle) - Copie" -> "Elegoo Neptune 4 (0.4 nozzle) - Copie"
I check "User preset" and validate

### Actual results

[image]

Orca slicer do not answer

### Expected results

Save my setting with the correct name 

### Project file & Debug log uploads

Log file
[debug_Tue_Jan_02_14_57_26_30096.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13810576/debug_Tue_Jan_02_14_57_26_30096.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

greendog99, 2024-01-08:
Same here, with the non-beta 1.9.0, on MacOS Sonoma:

Printer Settings -> Save Current Machine (floppy disc icon) -> Edit name of printer -> OK

Orca instantly crashes and the printer does not get renamed.

brendandebeasi, 2024-01-12:
Same problem here, MacOS Sonoma. 

Cannot export configs, instant crash when saving print settings...

brendandebeasi, 2024-01-13:
Uploading my crash log 
[OrcaSlicer-2024-01-12-140919.ips.zip](https://github.com/SoftFever/OrcaSlicer/files/13928892/OrcaSlicer-2024-01-12-140919.ips.zip)

brendandebeasi, 2024-01-13:
[OrcaSlicer-2024-01-12-140607.ips.zip](https://github.com/SoftFever/OrcaSlicer/files/13928898/OrcaSlicer-2024-01-12-140607.ips.zip)
Another one

eMadman, 2024-02-01:
Experiencing the same issue here

Kalus1, 2024-02-03:
Same issue here... v1.9.0 Windows

greendog99, 2024-02-21:
FYI this crash no longer happens for me in v1.9.1 (MacOS).

DiegoTheWolf, 2024-02-23:
Upgraded to 1.9.1 (Win10) still get those crashes.

Kalus1, 2024-02-23:
Upgrading to v1.9.1 seems to have fixed the issue for me. I'm on Windows 11.

jumika, 2024-02-24:
I still have this problem on v1.9.1

Daniel-Bartle, 2024-02-26:
I haven't been able to find a solution yet, but I think I found a workaround:
As long as you don't change the name of the printer, it seems to save any changed settings without crashing when you click save. To change the name of the printer I opened the file location where the printer info is stored and renamed both the .info and .json files to what I wanted the printer's name to be. Restarting Orcaslicer then applied the name change.
For Windows Installs the file location is "C:\Users\<user>\AppData\Roaming\OrcaSlicer\user\default\machine"

OldRhino, 2024-03-01:
I had it work and fail in v1.9.1.

The difference is this: if I edit a printer profile and just change the name and press save all is good.
                                   if I edit a printer profile, change a setting and name, press save - get hang when I close the dialog.
Hope this helps.

OldRhino, 2024-03-01:
PS> Once I save under a new name, I then can subsequently edit and save the new profile.

closed 2024-06-07 by github-actions[bot]